OrcaSlicer/OrcaSlicer #5284: Elephant foot compensation work in wrong direction.

Various699, 2024-05-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0/2.1.0-dev

### Operating System (OS)

Windows

### OS Version

Windows 11 Home

### Printer

BLV

### How to reproduce

Set elephant foot compensation for lets say 0.4

### Actual results

First layer should be thinner for set ammount but is wider, creating lip to the outside of model.
Test model have 1.2mm wall thickness and is hollow, in result first layer with elephant foot 0.4 is 1.56mm
[image]

### Expected results

First layer should be thinnker to lets say 1.0mm (without elephant foot enabled first layer have 1.3mm)
[image]

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

dstulken, 2024-05-06:
I can confirm that something odd is going on with the elephant foot compensation.
macOS Sonoma 14.4.1, OrcaSlicer v2.1.0-Dev (2024-04-28 nightly).

If I load the 'Voron 2.4 350 0.6 nozzle', add the cylinder primitive, and slice, it looks fine (straight walls).
If I change Elephant foot compesation from 0mm to 1mm and re-slice, the first layer is correctly inset.
If I change the Elephant foot compensation layers from 1 to 3, the first three layers are correctly (progressively) inset.

However, if I then add a cylinder primitive as a negative part, scaling that cylinder (uniform scale unchecked) to X = 33, Y = 33, Z = 35, then repositioning it to 0, 0, 0, things are not right.  (note: also disable the brim)   It appears as though the elephant foot compensation is not applied, even when changing the value (between 0.1 and 1mm) or the number of layers (1 or 3).  Checking the line width in the render pane confirms that the line width is unchanged.

Now, if we go back to that negative part, and change the size to 31, 31, 35 to thicken the walls, then the compensation *IS* applied correctly in that range of adjustments.

I cannot reproduce it, but at one point I also had an odd artifact where the first layer was the original size, then the *second* layer applied the compensation, and the third and above layers tapered back to the original size again.   I've tried to reproduce, but can't get it to reoccur.

dstulken, 2024-05-06:
> I cannot reproduce it, but at one point I also had an odd artifact where the first layer was the original size, then the second layer applied the compensation, and the third and above layers tapered back to the original size again. I've tried to reproduce, but can't get it to reoccur.

I figured it out.

Load the "Prusa MINIIS 0.4 nozzle" printer preset.   Add the cylinder primitive (18x18x18).   Add the negative modifier, and scale that cylinder to 16x16x18, and move it back to 0,0,0.   Set the elephant foot compensation to 1mm, and 5 layers.   This is the result.  First layer at the original non-adjusted size, second layer is inset, third-fourth-fifth layers appear identical to second (not tapering), and the sixth layer is back to the original size.

[image]

jahnj0584, 2024-05-06:
At first thought, a negative value should be used to shrink the first layer of the part, but considering its asking for a "compensation" amount of something that causes growth it also makes sense to say that a +0.2mm compensation would make it _smaller_

closed 2024-08-15 by github-actions[bot]

vadixidav, 2024-11-26:
@SoftFever Please reopen this issue as the bug still occurs as per my post above. It seems to have been marked stale by the bot.