OrcaSlicer/OrcaSlicer #5287: 2.1.0-dev Bambu X1C Not Seeing Camera

ranovr, 2024-05-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-dev

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambu Lab X1C

### How to reproduce

Try to connect to the video stream of the camera.

### Actual results

Orca cannot connect to camera with this build. It will say there is a network issue. If I roll back, the issue goes away.

### Expected results

I should be able to view the camera stream in the application

### Project file & Debug log uploads

No file needed. Refuses to connect regardless of project.
[J_Elites_1_Rifle_Combined.zip](https://github.com/SoftFever/OrcaSlicer/files/15224922/J_Elites_1_Rifle_Combined.zip)
[debug_Mon_May_06_08_32_17_48332.txt](https://github.com/SoftFever/OrcaSlicer/files/15224957/debug_Mon_May_06_08_32_17_48332.txt)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

-

KD7LRJ, 2024-05-08:
Seems like a duplicate of #5170

DivineJimmi, 2024-06-02:
It looks like OrcaSlicer has some problem with its own network-plugins.

Install the latest Bambu Studio, start it and let it install its latest network-plugins. After installation close Bambu Studio and open OrcaSlicer. It should work now.

mydevpeeps, 2024-07-04:
@DivineJimmi  confirmed here as well. still a problem in GA release of 2.1.1.
https://github.com/SoftFever/OrcaSlicer/issues/5754#issuecomment-2209553656

closed 2024-10-11 by github-actions[bot]