OrcaSlicer/OrcaSlicer #5289: Walls printing order does not work anymore

Z18Aplha, 2024-05-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Load STL file
2. Set "Walls Printing Order" to "Outer/Inner"
3. Slice
4. Check for random layers which walls are printed first
5. Change "Walls Printin Order" to "Inner/Outer"
6. Slice
7. Ceck for samel layers which walls are printed first

### Actual results

It makes no difference, which setting is chosen. In both cases the outer walls are printed first.

### Expected results

For "Walls printing order" "Inner/Outer" the inner walls should be printed before the outer walls.

### Project file & Debug log uploads

[debug_Tue_Apr_30_18_42_02_20400.log.zip](https://github.com/SoftFever/OrcaSlicer/files/15227650/debug_Tue_Apr_30_18_42_02_20400.log.zip)
[WallPrinting.zip](https://github.com/SoftFever/OrcaSlicer/files/15227653/WallPrinting.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

kisslorand, 2024-05-07:
I confirm this behaviour with the caveat that I could replicate it only a few times in many attempts (the included project was used). My multiple attempts were made by closing and reopening Orca Slicer.

rfnovo, 2024-05-13:
I'm seeing the same issue. Inner/Outer works fine, but everything else it get's wrong. Image attached. Also, every time it slices it does a slightly different wall order. 
[image]

mtc1122, 2024-05-27:
I'm able to reproduce this with inner/outer/inner. You can even use a primitive to test this.

No matter which model I use, inner/outer and outer/inner work correctly. However, it's on the first layer that inner/outer/inner seems to always slice as inner/outer. Wall count doesn't seem to affect this. Elephant foot compensation doesn't seem to affect it. I've tried many settings and cannot find one that changes this. It always slices incorrectly on the first layer.

To reproduce:
1. Place a cube
2. Set "Walls printing order" to inner/outer/inner
3. Preview the model. I find that all layers except for the first slice correctly. The first slices as inner/outer/inner.

closed 2024-09-02 by github-actions[bot]