OrcaSlicer/OrcaSlicer #5297: ARC fitting causing MCU TIMER TOO CLOSE

loutwice, 2024-05-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

Flsun v400

### How to reproduce

print multimaterial with arc fitting enable

### Actual results

crash in MCU TIMER TOO CLOSE always on the prime tower 

### Expected results

should not crash - issue already reported here https://github.com/SoftFever/OrcaSlicer/issues/1052

### Project file & Debug log uploads

[klippy (7).log](https://github.com/SoftFever/OrcaSlicer/files/15241117/klippy.7.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

printing without arc fitting enable works , the print is completed

igiannakas, 2024-05-09:
Reduce your arc resolution in klipper. Your SBC is too slow to handle higher resolution.

loutwice, 2024-05-09:
> Reduce your arc resolution in klipper. Your SBC is too slow to handle higher resolution.

@igiannakas already at 0.2 ,

igiannakas, 2024-05-09:
> > Reduce your arc resolution in klipper. Your SBC is too slow to handle higher resolution.
> 
> @igiannakas already at 0.2 ,

Prime tower btw shouldn’t be using arc commands. Do you have arc fitting turned off in the quality settings? If not, why not? Klipper has no benefit with it on, as it breaks the arcs down into lines anyway - instead with it on you’re actually loosing print quality as the model gets arc fitted and then broken down again in a lossy process.

the only place for a klipper printer to use arc commands is spiral z hop. For which a resolution  of 0.2-0.3 should be sufficient to prevent timer too close errors.

loutwice, 2024-05-09:
> Prime tower btw shouldn’t be using arc commands. Do you have arc fitting turned off in the quality settings? If not, why not? Klipper has no benefit with it on, as it breaks the arcs down into lines [...]

oh i didnt know ! thank you for your explanation!  so best to deactivate it ...  gotcha !

Ergonomicmike, 2024-05-12:
> Klipper has no benefit with it on, as it breaks the arcs down into lines anyway - instead with it on you’re actually loosing print quality as the model gets arc fitted and then broken down again in a lossy process.

@igiannakas I didn't know this either. It would be helpful if the tooltip said that Arc Fitting is "incompatible" with Klipper.

(I would offer to offer a PR. But I'm not a coder.)

igiannakas, 2024-05-12:
It’s not that it’s incompatible, it’s more that it’s pointless as klipper splits the arcs back out to segments (lines) using the resolution setting that you configure in the printer cfg file. 

I’ll make a tooltip adjustment, no worries ;)

Ergonomicmike, 2024-05-13:
@igiannakas That was fast. Thanks! I was going to ask if you knew of any other settings in Orca that also should be disabled when using Klipper, and if you could update the pop ups for them too.

(For when all the bugs in Orca have been squashed, the final ergonomic solution for this is to have Orca gray out settings like Arc Fitting when Klipper is set as the G-Code flavor for a printer.)

igiannakas, 2024-05-13:
No other setting has an adverse effect on Klipper from what I recall. This one specifically is a peculiar one - Marlin and Bambu benefit immensely from this being turned on as their motion planners take advantage of this. Klipper on the other hand doesnt.

Personally I dont have any other material deviation between my BBL profiles and my Voron profiles besides this setting.

PR #5352 (2024-05-18): Updated arc fitting pop up to clarify that this option should not be enabled for Klipper machines
# Description

Klipper does not natively support arc commands at the motion controller level. Hence enabling arc fitting on Klipper is detrimental to print quality as line segments are converted to arcs in the slicer and back to line segments in Klipper.

The revised pop up wording clarifies that this option is not recommended for Klipper printers. 

Fixes #5297 
Fixes #4216
closed 2024-05-18 by SoftFever