OrcaSlicer/OrcaSlicer #5313: Plate X,Y,Z origin

Alkerion, 2024-05-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When we have multiple plates on a project the origin stay on first plate, that's a non sense and annoying.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Origin must belong and be on the bottom left corner of the selected plate, and not on the first plate of the project.

tlhintoq, 2024-05-26:
Given the multiple related but not identical reports for all the various ways this XYZ system is screwing people up - it really would be great if this got addressed. 

I hope its not something that falls by the wayside because coordinate system is inherited from the master project (BambuSlicer/PrusaSlicer/Whichever)

tlhintoq, 2024-08-25:
So many people have opened tickets for this same thing under various names and complaints.
Fixing it one time would close a lot but more importantly make a lot of users VERY happy

tlhintoq, 2024-11-24:
Still - really needs attention when so many people are begging for better coordinate system

spali, 2025-02-26:
/unstale

Felix14-v2, 2026-10-07:
> When we have multiple plates on a project the origin stay on first plate, that's a non sense and annoying

You can manually bind the rotation origin to the center of the screen by holding `Ctrl`:

https://github.com/user-attachments/assets/94ffe785-a5f3-4890-8ebf-9e012adc7268

Not a perfect solution, but at least something that helps.

closed 2026-10-07 by ianalexis