OrcaSlicer/OrcaSlicer #5315: Please change the Z coordinate location on parts

Alkerion, 2024-05-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

yes, the coordinate system is not logic at all, and really annoying to use.
Also affect model resizing up and down, a nightmare when we have assembly and want to just resize a part or place a part on plate.
When I’ve to add a modifier or a negative part I’m going mad how nearly impossible it is to place the part at the exct desired place.
Seems crazy that Z = 0 doesn't put the item on the plate.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Please put the Z at the bottom of the part, not in the middle of it.

Or at least let user decide where he want the Z position with a setting in Preferences.

Thanks in advance !

tlhintoq, 2024-05-24:
I've thought the same thing and thought it was just me - that I'd get used to it.
But yeah, any other slicer 0,0,0 is the lower left corner at the base.
If I place an object at 0,0,0 I expect it to be on the bed, in the home corner.

So I'd submit its not just the Z that needs to be moved, but XYZ all need to be adjust to bottom, lower, left like any other slicer.

Alkerion, 2024-05-25:
> I've thought the same thing and thought it was just me - that I'd get used to it. But yeah, any other slicer 0,0,0 is the lower left corner at the base. If I place an object at 0,0,0 I expect it to be on the bed, in the home corner. This measuring from center of object in all dimensions is wacky.
> 
> So I'd submit its not just the Z that needs to be moved, but XYZ all need to be adjust to bottom, lower, left like any other slicer.

Fully agree, check my other post regarding the X0 and Y0 location : https://github.com/SoftFever/OrcaSlicer/issues/5313

tjgittiup, 2024-07-10:
BUMP! Please fix this. I've been trying to figure out how to stack items for a year, and still never figured it out. I have some stuff painted in Orcaslicer that I want to stack with a layer of PETG between PLA objects.... and the coordinates as is, make no sense to me... but I didn't realize it was starting at the lower corner of the bed... whacky! wild! ... can someone teach us how to use it, as it, if it makes sense to use it this way? ... to me, if I'm stacking items, I would start Z at 0, then add height of the next object, and so on... Thanks for your help and/or feature requesting this one!

discip, 2024-07-17:
@Noisyfox
Would you mind having a look at this? 😊

cramsvik, 2024-08-05:
It should at least be possible to choose to have Z=0 when placing the model on the plate. Whenever a print fails and I want to print the remaining (top) part it is really cumbersome to have to calculate the proper Z coordinate to lower the part to. In Cura I could just set it to negative X, where X was the height of the printed part. In Orca however it must be manually calculated.

tlhintoq, 2024-08-06:
Absolutely.  An object 500mm tall, and you want to print it in two halves means you start the second part at 250.  Its not rocket science when the coordinate system makes sense.  If I want an object to start at the left edge, hey that should be 0 not a case of ...
_The thing is 320mm wide, middle therefore is 160, the bed is 400wide, so I need the X location to be.. oh wait, now what?  Is it 160... or is it 400-160... or is it (400-160)/2..._ 
Honestly I don't even know.  I drag it left until Orca complains the object is now off the bed then I nudge it right.
And things get worse from there.  If I have 4 things and they all need 1/4 of the horizontal space... Well, I'm hosed on that when it should be as simple as 0, 100, 200, 300

Paco777, 2024-09-01:
And it looks like when the object is scaled, it is even more of a nightmare to calculate the correct position.
Trying to place some negative disks at the bottom of some scaled objects (to embed hidden magnets in prints) is really, really hard.
Even more so because I have to toggle between preview/prepare views to check if the slight movement I made fits into the solid or not.
As for the coordinates of the parts, maybe a simple toggle between object coordinates / world coordinates would be enough.

hxpphx, 2024-09-22:
I think if anyone change what is the facto the standard of coordinates, at least you should give users the possibility to choose between the new one and the old one. Even better would be to be able to choose between the 27 possible points. (See image)
[image]

discip, 2024-12-27:
@vovodroid
@buzzhuzz
Would someone of you guys please have a look at this? 😊

vovodroid, 2024-12-28:
I'm not sure it's easy to change without overhaul existing coordinate and slicing system.

buzzhuzz, 2024-12-28:
I'd agree with @vovodroid regarding complexity of those potential changes.

Currently origin of the model is the center of its bounding box. All the extra parts, negative parts and modifiers positioned relatively to the model origin. Moving object origin to the base of the bounding box will affect all aspects of objects manipulations.

However, it should be possible to implement few helpers which may be sufficient in most cases. I see them similar to the 'alignment' tool of the TinkerCad. I've got familiar with it last spring when my son got interested in 3d modelling and was impressed by the ease of use of that tool compared to the full-featured CAD systems: https://www.tinkercad.com/blog/tinkertip-align-to-reference

Another idea which may help with precise objects positioning may be extra modifier key to slowdown object movement by 1/10 or even by 1/100.

vovodroid, 2024-12-28:
There is new "Assemble" tool in the latest version (still not released though):

[image]

It allows touching desired faces.

discip, 2024-12-28:
First of all, thank you both very much!

> There is new "Assemble" tool in the latest version (still not released though):

Great!
Basically, this is the solution! 👍

ps:
I'm on the latest dev builds by default. 😃

tlhintoq, 2024-12-29:
The stacking tool is cool, no doubt.  I think the terminology needs some work because you need to first "assemble" to parts, then use the new "assemble" tool.  Just describing to someone what to do already gets confusing.  Maybe the new tool could be "stack" so it isn't confused with "assemble".
> First you have to assemble two parts so they are one assembly
> Then you can stack one part on the other part
Just reads better, yes?
[image]

tlhintoq, 2024-12-29:
And while this really nice stacking tool goes a long way to solve one use case - and I do thank you for your efforts.

It doesn't address the initial complaint of at least 5 open threads - that having ZERO be the center of the object in all axis is just horrible user experience that nobody likes and everyone is begging for 0,0,0 to be the lower left just like it is in every other CAD program from Cura to Fusion to Lychee to MeshMixer and on and on.

Lister-Storm, 2025-03-23:
Oh thanks im not alone. Was thinking im to stupid to find the option to change it, BUT there is no way to change it. It drives me crazy i cant set the hight to 0. Same if i scale a object hight its expending up and down instead of only upside. Its hell sometimes to position parts. Please change it to what the most other Slicer/Programms have it. 

But 10 month later nothing changed. Not really promising for the future. :(

tlhintoq, 2025-03-25:
https://github.com/SoftFever/OrcaSlicer/issues/5577

Zogar89, 2025-03-26:
Please. Fix This!

spali, 2025-03-26:
> > Please. Fix This! [@Noisyfox](https://github.com/Noisyfox) Prusa has world and local coordinate sistem. We can port that?
> 
> Yes. This will be the next thing I'll be doing if no other critical bug fixes needed.

Maybe back to the roots like prusa and allow to select. Prusa offers relative and world coordinates... Maybe an additional combination of these two settings with the option to zero on bottom left front.