OrcaSlicer/OrcaSlicer #5316: Remove any filament, not only the last one

Alkerion, 2024-05-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When we have a multicolored models and we want to remove a color that isn’t the last one it’s not possible without loosing the color on the model and that’s really annoying

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Let us remove the filament we want in the list instead of only the last one

Alpha12o, 2024-07-01:
Upvote. Would be great to add this option

closed 2024-10-07 by github-actions[bot]

Xeldin, 2025-11-06:
Please implement, would be extremely useful