OrcaSlicer/OrcaSlicer #5318: Assign fuzzy skin to a specific color

Alkerion, 2024-05-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

fuzzy skin is all or nothing, or require usage of modifiers, that's not very practical.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Will be nice to be able to assign fuzzy skin to as specific color.

That way we will be able to create surface effect exactly where we want by just painting objects.

vgdh, 2024-05-10:
Would be great to have some surface selection tool to apply fuzzy skin.

gogsi1, 2024-05-22:
I would like to see several features such as support for "spheres", "Fills" and adding only internal perimeters.

**bug**
If the outer contour is selected so that it does not pass into the inner contour, if there are holes on the model

ph0tic, 2024-07-21:
we seriously need this. if we can "apply by painting" at least it would solve some of this issue.

closed 2024-10-28 by github-actions[bot]