OrcaSlicer/OrcaSlicer #5332: Changing printer profile resets all filament, colors and process to last remembered ones for selected printer profile

r0mar0ma, 2024-05-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambulab X1C

### How to reproduce

1. Open any .3mf project with multiple filaments and color paintings and printer X1C
2. Switch to another printer profile P1S

[image]
[image]
[image]
[image]

### Actual results

After switched to another printer profile:
- ALL selected filament profiles resets to last ones that was used before with new selected P1S profile
- ALL coloring removed
- process switched to another one

### Expected results

Should not associate, remember and reset filament profiles and process profiles when changing printer profile, like it's working in BambuStudio.
Should not reset coloring.

### Project file & Debug log uploads

[Cylinder.zip](https://github.com/SoftFever/OrcaSlicer/files/15282380/Cylinder.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

There is no any way to print multimaterial files with coloring because changing of printer profile from the one which is saved into the project to the one which I want to use just resets all filaments and colorings.

BambuStudio is not remembering "last used" filaments and process for the printer which is much more correct way.

If this remembering is "the feature" then probably should be the an option to disable this. Current workflow with remembering is making OrcaSlicer unusable to print such projects.

pennystockss, 2024-06-29:
Yes absolutely yes need this feature. Changing between printers is super frustrating when we have to change all the colors, let alone if new printer doesn't have enough colors need to add them, then reload 3mf file. If you forget to do that then save you erased your painting.

rdbischoff, 2024-06-29:
pointless to share 3MFs without being able to switch to your printer without keeping the color painting and such.

witchrobin, 2024-07-03:
I also ran into the same problem when trying to open a multicolor print from Makerworld where it would reset all the filament from the 3mf file once I change printer.  Found out there is an option in preferences to "Remember printer configuration".  Turning that off fixed the problem and behaves like BS.

L2on1, 2024-07-03:
dub of #4145 it explains well the same issue with painting for instance and #5446 , #3898

kemper, 2024-09-06:
I've mostly been using Orca since getting a Bambu P1S and thought that model creators were just doing something dumb with their 3MF files. Then I decided to google a bit and realized this happens with Orca and not Bambu Studio. Glad I have a workaround but this means I'm probably not going to use Orca since I download so many 3mf files.

slooker, 2024-10-18:
I'm having the same problem, unfortunately.  If I have to change the printer choice, it drops all colors and assigns random colors.  I have "Remember printer configuration" checked.  
[image]
[image]
[image]

CaptianNoob, 2024-12-01:
> I also ran into the same problem when trying to open a multicolor print from Makerworld where it would reset all the filament from the 3mf file once I change printer. Found out there is an option in [...]

This is exactly what I needed. Thank you good sir.

Lomax21, 2025-01-26:
> I also ran into the same problem when trying to open a multicolor print from Makerworld where it would reset all the filament from the 3mf file once I change printer. Found out there is an option in [...]

This 100% solved the issue... I was having to open files in BS to switch from A1 to my P and X's just to save the profile settings much less the filament colors. After unchecking that option, it works like it should now.

vinny1575, 2025-03-10:
> I also ran into the same problem when trying to open a multicolor print from Makerworld where it would reset all the filament from the 3mf file once I change printer. Found out there is an option in [...]

Thank you for this!

adfaklsdjf, 2025-06-13:
does that mean this is the intended behavior?  it's hard to imagine why i'd want to use "Remember printer configuration" setting.  

@witchrobin you're a lifesaver.

adfaklsdjf, 2025-11-09:
Still seems like a bug to me... ¯\_(ツ)_/¯

kemper, 2025-11-29:
I used gemini to research and created a proposed fix. I haven't been able to successfully build the project to test it, but the change seems reasonable.

M2ABRAMSTANK, 2026-08-19:
Same issue here, disabling remember printer configuration avoids the reset, works correctly in Bambu Studio either way. Same on both latest nightly & stable