OrcaSlicer/OrcaSlicer #5336: RatRig printer profile has wrong pause command.

valentt, 2024-05-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Windows, Linux, Mac

### Printer

RatRig V-Core 500 (But any klipper based machine will have same issue)

### How to reproduce

Add pause in OrcaSlicer, slicer adds M601 command which is not know by Klipper, so it just skips it.
[image]

Klipper uses "PAUSE" as pause command.

### Actual results

Machine doesn't pause.

### Expected results

Machine should pause.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I created pull request to fix this issue:
https://github.com/SoftFever/OrcaSlicer/pull/5335

xBoKoBx, 2024-05-12:
Same issue for FLSUN V400..

cochcoder, 2024-05-26:
> Same issue for FLSUN V400..

This should be fixed with PR #5469

closed 2024-05-28 by SoftFever