OrcaSlicer/OrcaSlicer #5340: OrcaSlicer crashing intermittently when slicing with Tree Hybrid supports enabled

AdminChobe, 2024-05-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Windows 10 Home 22H2

### Additional system information

CPU: Intel i7-12700
RAM: 32GB 3200MHz

### Printer

Creality Ender 3

### How to reproduce

1. Select Tree Hybrid Supports with supports enabled and Automatic tree supports selected.
2. Begin slicing model
3. Program will usually (~90% of attempts) crash with 'Divide by Zero' error.

### Actual results

Program will close after briefly displaying 'Divide by Zero' error.
[image]

### Expected results

Model should slice successfully with Hybrid Tree supports.

### Project file & Debug log uploads

[Organic Support Test.zip](https://github.com/SoftFever/OrcaSlicer/files/15285222/Organic.Support.Test.zip)
[crash_Sun_May_12_16_06_00_0.zip](https://github.com/SoftFever/OrcaSlicer/files/15285234/crash_Sun_May_12_16_06_00_0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Similar issue has been reported on previous versions and has been marked as patched, yet persists. Does not occur when using any other type of tree supports.

3LD, 2024-05-16:
I'm getting this error on all tree styles in 2.0. 
occasionally works, but 9/10 fails with 'a fatal error occurred: "Divide by zero"'

barsamgr, 2024-07-17:
I can confirm this bug on 2.1.1 with Sonoma 14.5

closed 2024-10-23 by github-actions[bot]