OrcaSlicer/OrcaSlicer #5343: Skirt minimum extrusion length like superslicer

Thetraktor01, 2024-05-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When i mounted an ERCF v2 to my 2.4, my normal purge line no longer worked, because of extruder sync. Therfore i began using skirts again.
My ERCF is set up to unload filament after each print, with a cutter, so a small piece of filament is left in the hotend. When starting a new print with a different filament, the print is not the right colour in the beginning so i purge into a skirt.
The problem is that with a small print the skirt is too small to properly purge the filament. The solution to that is up the amount of skirt lines. That however is a problem with bigger prints, because then it wastes too much time and filament on skirt, if i dont manually change the amount of skirt lines manually.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The feature should have a minimum purge amount like SuperSlicer, where it makes the skirt bigger on small prints, and smaller on large prints to reduce waste
The feature in superslicer is called "Minimal filament extrusion length" in SuperSlicer

### Additional context

[image]

netweaver1970, 2024-05-12:
This has been asked before. But there were always juicier things to work on, it seems. Maybe this time we can get some traction.

igiannakas, 2024-05-13:
I've pushed a PR above that adds this parameter. Indeed it can be useful instead of a purge line.

closed 2024-05-20 by SoftFever

tonycstech, 2025-05-12:
Absolutely agree with that.
Cant use BRIM. I need to extrude the skirt so small prints dont run dry.
If i print 10x10 cube, skirt does not get printed because there is no filament inside the nozzle yet.
Only when cube starts to print, about half way on the 1st layer filament starts to come out.

The option for “minimum extrusion length” must be there.
If you print large object, there is no need to print entire skirt since its too large. Instead just print the minimum extrusion length and be happy but no. China decided its not needed.