OrcaSlicer/OrcaSlicer #5345: Unable to create filament

wKich, 2024-05-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

MacOS 14.2.1

### Printer

Creality K1

### How to reproduce

1. Click "Set filaments to use" cog button
2. Choose "Custom filaments" tab
3. Click "Create New" button
4. Choose Eryone -> PETG-CF -> Basic Filament
5. Under "Add Filament Preset under this filament" section, there is no options to choose
6. Click "Create" and see error message "You have not selected a printer or preset yet. Please select at least one"

### Actual results

In "Create filament" window, because of empty presets it's impossible to add filament and you stuck with an error "You have not selected a printer or preset yet. Please select at least one"

### Expected results

I see two options:
- Get more user friendly message, explaining what to do if there is no presets available. So at least I don't know how to add custom user's presets
- Allow to choose more general presets. For example, I have Creality K1 printer and I only see Creality filaments presets and for unknown reason, I don't see more basic filaments and it seems they are considered as incompatible(!!!WTF) with Creality K1

### Project file & Debug log uploads

I only have one added printer, Creality K1. So I think it's reproducible in fresh OrcaSlicer setup

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2024-05-14:
Duplicate of this bug from 2023 that still hasn't been addressed.
https://github.com/SoftFever/OrcaSlicer/issues/3242

wKich, 2024-05-15:
@tlhintoq you are right. I'll close my issue

closed 2024-05-15 by wKich