OrcaSlicer/OrcaSlicer #5347: Multiple objects same build plate

BiggumBuddha, 2024-05-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm trying to print something that requires 3 colors, and base plate and frame and embossed inside is a character outline in white with black pupils. I can't find any support on doing this with an Ender 3 v2. Is there a repository with this information or is it in the works?

### Which printers will be beneficial to this feature?

Klipper, Marlin

### Describe the solution you'd like

Configure multiple color prints on one plate and not by layer

vgdh, 2024-05-12:
You can enable manual filament change in printer settings.
Though it will be manual labor intensive print.

BiggumBuddha, 2024-05-12:
I initially thought that as well, but my Ender 3 v2 didn’t register the gcode from the slicer to stop for the change. Sent by the squirrels in a box

igiannakas, 2024-05-13:
This is not a slicer bug. Please close issue down and seek support in the discord server.

igiannakas, 2024-05-14:
You can set  pause command in the filament change gcode box in the printer settings. Also enable manual filament change so the Tx commands are not emitted.

igiannakas, 2024-05-14:
See below pause command emitted to gcode when setup for manual filament change + the change filament g-code is PAUSE.

This is a setup issue, not a slicer bug...

[image]

[image]

[image]

BiggumBuddha, 2024-05-14:
I would agree, except T1 isn’t a supported command on Ender machines. Not having a valid code option (in my mind) makes that a slicer bug, as the expectation is that the gcode commands are expected to work from the slicer and the slicer is compatible with the Ender series. Sent by the squirrels in a box

image.jpeg (view on web)
image.jpeg (view on web)
image.jpeg (view on web)

BiggumBuddha, 2024-05-14:
> If you enable manual filament change, the Tx commands are not issued in
> the gcode. See my screenshot above... What you see in the screenshot is a
> GCODE comment ;)

Right, and that was my setup before I submitted the ticket (Manual filament
change enabled), and it issued the T1 command, which is ignored by Ender
series. What should have been a MMU (3) print was a single with no pause
for filament change with manual enabled.

igiannakas, 2024-05-14:
In the gcode above I don’t see a Tx command but rather the pause command where the tool changes. Implement the changes as exactly I’ve screenshotted them above. If it doesn’t work post the project file as without it I cannot see what you’re seeing…

BiggumBuddha, 2024-05-14:
Right on. I most likely didn't, as I assumed enabling would augment the
PAUSE command. I'll give it a run today, and update afterwards.

closed 2024-08-21 by github-actions[bot]