OrcaSlicer/OrcaSlicer #5355: travel speed are overwrited by small perimeter speed

gabwhoop, 2024-05-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

ender 3 custom

### How to reproduce

small perimeter = 12mm
travel speed = 200 mm

### Actual results

travel speed is overwritten by small perimeter speed in small perimeters.....

[image]
[image]

### Expected results

better quality 
no strings 
reduce printing time

### Project file & Debug log uploads

[small perimeter test.zip](https://github.com/SoftFever/OrcaSlicer/files/15294297/small.perimeter.test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Dconn88, 2024-06-17:
I can confirm this is an issue and I found this on commenting instead of putting my own. (Sorry I'm new to git hub, but not to 3-D printing!) Just wanted to try to help built such an awesome product!

My small parameters are set at 50% speed, and when it prints all the screw holes simultaneously, the travel between the holes is a very taking 50% of my travel speed.
For example what I noticed is travel speed is set at 500 mm/s but it traveling around 250 mm/s in between small parameters when they are one after another. The preview still shows 500 mm/s when you look at the speed of visual, but actual is half of that, just in between small parameters. Everything else is great. 
I'm guessing it is about 250 mm/s based on my internal perimeters at 300 mm/s and external at 200 mm/s for my visual reference/calibration of eyesight, so that's why I am sure it's taking the 50% perimeter speed, and @gabwhoop's post confirms what I was thinking.

On the screenshot, shows 500 mm/s for all travel speed, but every travel between small perimeters ( screw holes) is visibly about half speed. Other travel is definitely 500 mm/s.

Thank you so much! Keep Up the good work!
[image]

Dconn88, 2024-06-19:
Also, the same thing happens with travel speed on the 1st layer as well. My 1st layer travel speed is set at 50%, and my 1st layer speed is set at 50, 1st layer infill at 105. It should be doing 250 mm/s, but actual travel speed is about 50 mm/s. I think it's just taking the 50 from the initial layer speed.

closed 2024-09-25 by github-actions[bot]