OrcaSlicer/OrcaSlicer #5359: Solid Top Surfaces get weird random edges

TheAtomicSoul, 2024-05-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

na

### Printer

Bambu P1P

### How to reproduce

Trying to print this item https://makerworld.com/en/models/461094?fbclid=IwZXh0bgNhZW0CMTAAAR0WGmZmWQ5koJEwpNQlygEBsxaXVhtHudnEkFOM9vUeJnoowc_SpM0B1Ww_aem_Af1cC7HpivL5PTd1EGxk362ThmWvD_hP2DnmcDMVI2K8rxRtviKZE7Dg4J4YEE86pm2w7lmofhhXxPmNrun8nAQB#profileId-369772

and am trying to make the top blue surface smooth, but the surface is getting random lines in it, rather than a smooth surface, or even one following the text that goes on top of it. 

[image]
[image]

### Actual results

gets weird lines in the surface

### Expected results

Get a solid flat surface without the weird lines. 

### Project file & Debug log uploads

[BlockBuster_Magnet_Keychain Test.zip](https://github.com/SoftFever/OrcaSlicer/files/15301614/BlockBuster_Magnet_Keychain.Test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

na

slothy89, 2024-05-21:
You need to look at the One-Wall threshold setting on teh Quality tab. Set this to 100% and the lines disappear.

The setting exists partly for this reason, and even has a tooltip explaining that higher than 100% will likely cause these exact artifacts.

This is not an unknown limiation, and is a setup error not a bug.

TheAtomicSoul, 2024-05-21:
> You need to look at the One-Wall threshold setting on the Quality tab. Set this to 100% and the lines disappear...

Thank you, I did not know about that setting!

closed 2024-05-21 by TheAtomicSoul