OrcaSlicer/OrcaSlicer #5369: Orca Slicer - Arch Linux - orca-slicer-bin

4e576rt8uh9ij9okp, 2024-05-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

latest (v2.0) github

### Operating System (OS)

Linux

### OS Version

Arch Linux x86_64

### Additional system information

OS: Arch Linux x86_64 
Host: NUC13ANHi3 M89901-203 
Kernel: 6.8.9-arch1-2 
Uptime: 1 day, 22 hours, 28 mins 
Packages: 763 (pacman), 13 (flatpak) 
Shell: bash 5.2.26 
Resolution: 3840x1600 
WM: sway 
Theme: Adwaita [GTK2/3] 
Icons: Adwaita [GTK2/3] 
Terminal: foot 
CPU: 13th Gen Intel i3-1315U (8) @ 4.500GHz 
GPU: Intel Raptor Lake-P [UHD Graphics] 
Memory: 5450MiB / 15516MiB 

### Printer

Custom Manta M8P + CB1 + Klipper + Mainsail

### How to reproduce

1. rua install orca-slicer-bin
2. orca-slicer

### Actual results

```
[user@ArchPC ~]$ orca-slicer
[2024-05-13 20:46:58.304432] [0x00007b0a9c64e480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:28435): GLib-GObject-CRITICAL **: 20:46:58.491: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.491: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:28435): GLib-GObject-CRITICAL **: 20:46:58.491: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.491: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:28435): GLib-GObject-CRITICAL **: 20:46:58.492: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.492: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:28435): GLib-GObject-CRITICAL **: 20:46:58.493: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.493: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:28435): GLib-GObject-CRITICAL **: 20:46:58.493: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.493: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:28435): GLib-GObject-CRITICAL **: 20:46:58.494: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.494: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:28435): GLib-GObject-CRITICAL **: 20:46:58.494: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.494: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:28435): GLib-GObject-CRITICAL **: 20:46:58.495: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.495: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:28435): GLib-GObject-CRITICAL **: 20:46:58.496: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.496: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:28435): GLib-GObject-CRITICAL **: 20:46:58.496: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.496: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.571: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.575: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.575: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.575: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.575: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.575: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.575: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.575: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.575: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.580: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.580: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.580: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.640: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.640: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.646: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.646: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.648: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.648: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.655: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.655: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.660: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.660: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.664: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.664: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.671: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.671: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.676: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:28435): Gtk-CRITICAL **: 20:46:58.676: gtk_widget_set_size_request: assertion 'width >= -1' failed
Segmentation fault (core dumped)
```

### Expected results

Start the slicer

### Project file & Debug log uploads

Look actual results

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I run on Sway Wayland, maybe this is an issue.

strap-9021, 2024-05-15:
I don't think Wayland is the problem I'm using x11 with kde 6.0.4 and it gives me the same error

4e576rt8uh9ij9okp, 2024-05-15:
> I don't think Wayland is the problem I'm using x11 with kde 6.0.4 and it gives me the same error. I also tried running Appimage but it also gives the same errors
> ## **this is the result of valgrind**
> 
> ==27306== Invalid read of size 8 ==27306== at 0x1F00069: wxMediaCtrl2::wxMediaCtrl2(wxWindow*) (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x1E7CDA1: Slic3r::GUI::StatusBasePanel::create_monitoring_page() (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x1E81484: Slic3r::GUI::StatusBasePanel::StatusBasePanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&) (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x1E8AA59: Slic3r::GUI::StatusPanel::StatusPanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&) (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x190956E: Slic3r::GUI::MonitorPanel::init_tabpanel() (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x1909D03: Slic3r::GUI::MonitorPanel::MonitorPanel(wxWindow*, int, wxPoint const&, wxSize const&, long) (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x16D18D5: Slic3r::GUI::MainFrame::init_tabpanel() (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x16D2F55: Slic3r::GUI::MainFrame::MainFrame() (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x1688DDB: Slic3r::GUI::GUI_App::on_init_inner() (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x168A1F9: Slic3r::GUI::GUI_App::OnInit() (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x41D8E26: wxEntry(int&, wchar_t**) (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x164BC1A: Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (in /opt/orca-slicer/bin/orca-slicer) ==27306== Address 0x18 is not stack'd, malloc'd or (recently) free'd ==27306== ==27306== ==27306== Process terminating with default action of signal 11 (SIGSEGV): dumping core ==27306== Access not within mapped region at address 0x18 ==27306== at 0x1F00069: wxMediaCtrl2::wxMediaCtrl2(wxWindow*) (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x1E7CDA1: Slic3r::GUI::StatusBasePanel::create_monitoring_page() (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x1E81484: Slic3r::GUI::StatusBasePanel::StatusBasePanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&) (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x1E8AA59: Slic3r::GUI::StatusPanel::StatusPanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&) (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x190956E: Slic3r::GUI::MonitorPanel::init_tabpanel() (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x1909D03: Slic3r::GUI::MonitorPanel::MonitorPanel(wxWindow*, int, wxPoint const&, wxSize const&, long) (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x16D18D5: Slic3r::GUI::MainFrame::init_tabpanel() (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x16D2F55: Slic3r::GUI::MainFrame::MainFrame() (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x1688DDB: Slic3r::GUI::GUI_App::on_init_inner() (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x168A1F9: Slic3r::GUI::GUI_App::OnInit() (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x41D8E26: wxEntry(int&, wchar_t**) (in /opt/orca-slicer/bin/orca-slicer) ==27306== by 0x164BC1A: Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (in /opt/orca-slicer/bin/orca-slicer) ==27306== If you believe this happened as a result of a stack ==27306== overflow in your program's main thread (unlikely but ==27306== possible), you can try to increase the size of the ==27306== main thread stack using the --main-stacksize= flag. ==27306== The main thread stack size used in this run was 8388608. ==27306== ==27306== HEAP SUMMARY: ==27306== in use at exit: 35,910,415 bytes in 389,967 blocks ==27306== total heap usage: 1,157,710 allocs, 767,743 frees, 128,835,219 bytes allocated ==27306== ==27306== LEAK SUMMARY: ==27306== definitely lost: 249,284 bytes in 1,345 blocks ==27306== indirectly lost: 334,677 bytes in 13,417 blocks ==27306== possibly lost: 402,718 bytes in 1,318 blocks ==27306== still reachable: 32,246,512 bytes in 353,197 blocks ==27306== of which reachable via heuristic: ==27306== newarray : 32,160 bytes in 26 blocks ==27306== multipleinheritance: 5,624 bytes in 42 blocks ==27306== suppressed: 368 bytes in 2 blocks ==27306== Rerun with --leak-check=full to see details of leaked memory ==27306== ==27306== For lists of detected and suppressed errors, rerun with: -s ==27306== ERROR SUMMARY: 2 errors from 1 contexts (suppressed: 0 from 0) Segmentation fault (core dumped)

I switched to Prusa Slicer
`sudo pacman -S prusa-slicer`
Nice thing is that it's a official package.

strap-9021, 2024-05-15:
ok I found a solution, try and see #5258, I solved it by installing `gst-plugins-base`

LucHeart, 2024-06-15:
> ok I found a solution, try and see #5258, I solved it by installing `gst-plugins-base`

+rep this is the solution.. would be nice if this was mentioned in the readme

EBADBEEF, 2024-06-17:
The segfault in wxMediaCtrl2 is due to null pointer access because wxMediaCtrl::Create() fails.

References:
https://github.com/SoftFever/OrcaSlicer/blob/v2.1.0-beta/src/slic3r/GUI/wxMediaCtrl2.cpp#L42
https://github.com/SoftFever/OrcaSlicer/blob/v2.1.0-beta/src/slic3r/GUI/StatusPanel.cpp#L1042

It crashes due to dereferencing m_impl (I think inherited from wxMediaCtrl class) even though it is null. Other places in the code have a guard to check m_impl is not nullptr, so I think that would probably be the right thing to do here. Also could check if Create() fails (I see it returning false).

Next step for me is to debug exactly why wxMediaCtrl::Create is failing, but it's almost certainly GStreamer related. I have the GStreamer libs installed and it's still failing (nixos 21.04 on x86_64 running the released appimage binary). I'm going to track it down shortly!

Ginkooo, 2024-06-30:
I had the same problem, but I already had `gst-plugins-base` installed. Turned out, that installing 32 bit version of it helped :)

serhii-nakon, 2024-07-15:
I use Debian 12 and OrcaSlicer segfault too while just clicking on random tabs without any changes inside this tab. I have installed gst base plugins

serhii-nakon, 2024-07-15:
Here logs 

```
serhy@AIML:~$ cat .config/OrcaSlicer/log/debug_Mon_Jul_15_17_37_47_54399.log.0 
[error]	2024-07-15 17:37:50.330310[Thread 0x00007f85c1485440]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2024-07-15 17:37:50.497447[Thread 0x00007f85c1485440]:05:37:50 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-07-15 17:37:50.501322[Thread 0x00007f85c1485440]:05:37:50 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-07-15 17:37:50.503142[Thread 0x00007f85c1485440]:05:37:50 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-07-15 17:37:50.515487[Thread 0x00007f85c1485440]:05:37:50 PM: Warning: Error running JavaScript: Unsupported result type
[error]	2024-07-15 17:37:51.313673[Thread 0x00007f85c1485440]:calc_exclude_triangles:Unable to create exclude triangles
serhy@AIML:~$
```

4e576rt8uh9ij9okp, 2024-07-20:
In my eyes this is a big problem because a decent amount of people is unable to use OrcaSlicer.
I hope that someone will take this bug and fix it... since May... nothing happen still doesn't work.

closed 2024-10-27 by github-actions[bot]