OrcaSlicer/OrcaSlicer #5376: Wall Loops number Override Bridge and cause failiure.

lhndo, 2024-05-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

LH Stinger

### How to reproduce

Slice model attached
Look at the bridging on the fan ducts.
The more perimeters you have, the worse it is. Works with only one 😹

### Actual results

Perimeters floating instead proper supported bridge lines  
[image]
[image]

### Expected results

Internal bridges are anchored properly. Perimeters not printed in air when the bridge can expand to a solid point underneath.

### Project file & Debug log uploads

[bridge error toolhead.zip](https://github.com/SoftFever/OrcaSlicer/files/15324435/bridge.error.toolhead.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lhndo, 2024-05-15:
Some more examples of perimeters "growing" into the bridge area
[image]
With one wall the bridge can actually bridge
[image]

lhndo, 2024-05-15:
[Impossible Cube.zip](https://github.com/SoftFever/OrcaSlicer/files/15326666/Impossible.Cube.zip)
[image]

Curious if anyone can make a successful bridge layer when setting the perimeters above 4 on this impossible model.

discip, 2024-05-16:
I know it's not a real solution, but as a temporary workaround you could change the **`Walls printing order`** to **`Outer/Inner`**.

buzzhuzz, 2024-05-17:
@lhndo, another workaround may be enabling "Bridge counterbore holes" to either partially bridged or sacrificial layer:
[image]

However, this might compromise part strength if your wall is too thin and bridge density (or flow) less than 100% as vertical wall will be fully covered by the bridge layer added:
[image]

lhndo, 2024-05-17:
@buzzhuzz that's a good trick! I managed to do something similar in SuperSlicer by using custom settings and bridge orientation. That only worked in one direction sadly. 
<br>
Unfortunately, for me, it doesn't apply in the original case. 
The main point for the latter example was to create a simple test case to illustrate the original issue, and maybe help the testing process if anyone has a go at tackling this issue. 
<br>
In my case I have a dozen variations of this toolhead, which is used for the LH Stinger, and are being printed by other people.

lhndo, 2024-05-19:
Another example of the same issue:
[image]

BlackNet, 2024-06-16:
Been dealing with this same issue for the past 2 weeks.

lhndo, 2024-08-25:
Another example: 
[image]

A possibly smarter implementation of "Remove perimeters on bridges" from SuperSlicer which would leave the anchored perimeters segments in place would be welcomed. 
[image]

closed 2024-12-01 by github-actions[bot]

andrewleek, 2025-08-30:
Howdy everyone,
I have made a PR regarding overhangs and bridges and it may solve some of the issues in this thread. In particular the 
tensioner_error example.

It would be great to get some testing and feedback to make sure it works correctly and there aren't any unforeseen side-effects.

PR:
https://github.com/SoftFever/OrcaSlicer/pull/10565

You can download the builds under "artifacts":
https://github.com/SoftFever/OrcaSlicer/actions/runs/17304272227?pr=10565

Thank you

lhndo, 2025-09-01:
Thanks a lot for looking into this @andrewleek! I'll give it a go for sure