OrcaSlicer/OrcaSlicer #5380: Ability to override nozzle size warning is missing

anselor, 2024-05-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1 nightly

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Device name	LAPTOP-K8C6P1BB
Processor	AMD Ryzen 7 5800U with Radeon Graphics            1.90 GHz
Installed RAM	16.0 GB (15.3 GB usable)
Device ID	63657387-77D0-4759-88CD-0A4F05404E5D
Product ID	00325-80000-00000-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

Bambu Lab P1S

### How to reproduce

Create a custom nozzle profile with 0.25mm diameter
Try to print

### Actual results

Get an error that the nozzle size doesn't match the printer configuration and am blocked from printing

### Expected results

In 1.9.1 there was an option to confirm and proceed to print anyway.

### Project file & Debug log uploads

[blob_emoji_6x3_b.zip](https://github.com/SoftFever/OrcaSlicer/files/15330542/blob_emoji_6x3_b.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

With the release of the Panda Revo a lot more nozzle diameters are available now that aren't 'standard'.

anselor, 2024-05-16:
I'm able to create custom nozzle profile inheriting from the 0.2mm nozzle:
[image]

In 1.9.1 and 2.0.0, when I print I'm presented with the following dialog. Clicking Confirm allows me to print anyway and it prints correctly:
[image]

In the 2.1.0 Nightly it instead gives me this dialog that blocks my progress:
[image]

Eldenroot, 2024-05-16:
I can confirm, annoying bug

@SoftFever

anselor, 2024-05-22:
The change came in from BambuStudio in commit [1bb8fad63fa4f4bcaeb039aec030b1247055c88a](https://github.com/SoftFever/OrcaSlicer/commit/1bb8fad63fa4f4bcaeb039aec030b1247055c88a)

I think just removing this line will restore the ability to override the nozzle diameter warning:
https://github.com/SoftFever/OrcaSlicer/blob/0a7f787e3081c3868bf105e2861ed087d4ce2c48/src/slic3r/GUI/SelectMachine.cpp#L2576C32-L2576C39

@SoftFever

anselor, 2024-05-26:
@Eldenroot My fix seems to work. Should be in whenever the next build happens.

closed 2024-05-26 by anselor

Elemecca, 2024-08-24:
This issue is also present in v2.1.1. It was re-introduced in 27f140fb1889ab6890aafd6c0035235cf5316772.

It was subsequently fixed in 238bf49f4769bc8e8813df3e9b4d4d1237c40b67 (#5900), which has not yet been released.