OrcaSlicer/OrcaSlicer #5383: When I Do Support Painting It Gets Stuck At 65%!

ysnuysl, 2024-05-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Custom - hypercube

### How to reproduce

When I slice any .stl file with the automatic support or manual support option selected, it can produce .gcode in a very short time without any problems. However, when I paint with the manual support option turned on, it slices up to 65%, then gets stuck and then gives an error.

### Actual results

As a result of waiting for a long time, an error occurs, or the program first freezes and then closes.

### Expected results

It should be able to perform the slicing process without any problems.

### Project file & Debug log uploads

[image]
[image]
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Sidiusz, 2024-07-23:
Yep, same

Kurios1969, 2024-10-14:
Same issue. Appears to have something to do with Hybrid Style supports. If I have hybrid and do Manual Tree supports it will freeze at 65% on anything complex (which is when I need manual painting). Switching one setting, Style Default, it completes as normal.

tlhintoq, 2024-12-29:
I've been fighting this for a long time.  One thing I learned somewhere in some post... @SoftFever mentioned that this style of support can't be split into multi-threaded tasks to be calculated in parallel.  Its all single-threaded.  So not able to take advantage of any of the raw horsepower of having a 12 CPU computer etc.  That's why it feels like you're driving an 486 CPU.

I've had jobs like this one (see below) appear stuck for over an hour... over an entire afternoon if the job is large (1 meter cubic printer).  But it does generally finish.  Eventually.

As soon as Orca is not the foremost application it is no longer priority processing.  If you stick it in the background to keep slicing while you do other things you'll just make it worse/slower/longer.  Yeah, that hurts.  Having a $3,400 powerhouse computer dedicated to slicing one job for 5 hours and can't do any other task.  Honestly, I make these big jobs that require this support choice the last job of the day, start it going, ensure that Orca is the only program running and was clicked on to ensure its front-most, then send it slicing overnight while I go home.  If I'm lucky its done in the morning when I come in.

[image]

closed 2025-04-07 by github-actions[bot]