OrcaSlicer/OrcaSlicer #5384: Initial Layer Part Cooling Fan Speed

JeanKevyn, 2024-05-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I bought a Creality K1 9 months ago. Since the first few weeks with the printer, it has always had a constant clogging problem. Today, several months later, I finally understood the reason for these clogs. Basically, the Creality K1 clogs because the part cooling fan NEVER activates on the first layer, significantly heating the hot-end and preventing filament flow. I managed to solve the problem because dozens of people were experiencing the same issue and reported that simply setting the part cooling fan to 30% for the first layer solves the problem. Indeed, this strategy worked perfectly. However, the option to configure the part cooling fan speed on the first layer is not available. I tried working with the Max and Min Threshold parameters but did not achieve the desired results. Basically, all I need is for the first layer to have the part cooling fan at 30% speed, and ALL other layers to have the part cooling fan at 100%.

Is there a way to achieve this goal? It could even be a solution via G-Code. I just want to stop having this constant clogging headache.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

I would just like the option to activate this specific fan at a specific speed on the first layer.

### Additional context

[image]

vgdh, 2024-05-17:
I familiar with your problem, I use petg fan air duct and it get melt if no air flow provided all the time. 
I use KLIPPER and if the part is big enough I have to enable fan manually in mainsail web panel.

What is interesting that some times it don't sets speed to zero for the first layer and leaves the same speed as it has been before the print started (probably it assumed it already should be zero). 

I think the first layer fan speed will be very handy feature.

buzzhuzz, 2024-05-20:
There are actually special case just for your configuration:
https://github.com/SoftFever/OrcaSlicer/blob/79ba04515093cc7350834dd891f783123e0d2169/src/libslic3r/GCode/CoolingBuffer.cpp#L748-L752

Which is literally forcing zero cooling for the first layer if gradual cooling ramp up is enabled.
You may try set "full fan speed at layer 0" to 0 manually for bigger prints only. Assuming small footprint objects will not cause nozzle overheat anyways.

Proper solution I see is to add something like "first layer cooling speed" which also will act as a starting point for cooling speed ramp up during "full fan speed at layer" layers.

closed 2024-08-27 by github-actions[bot]

GustawXYZ, 2025-07-23:
It would be really nice if this could be implented as it indeed causes heat creep for me too. Setting the fan to 10% for the first layer solves the problem. For now I added a gcode macro that sets a min part fan variable and then modified M106 macro to always make sure speed is above min_part_fan value.