OrcaSlicer/OrcaSlicer #5388: Printer-Speed values don't all show up

inspectionsbybob, 2024-05-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Win11 Latest patch

### Additional system information

Device name	IBB-HP600
Processor	Intel(R) Core(TM) i7-6700 CPU @ 3.40GHz   3.41 GHz
Installed RAM	32.0 GB (31.9 GB usable)
Device ID	D2EDBE95-2141-4DF7-BE8E-D2984008F7CE
Product ID	00330-80000-00000-AA552
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

Ender3 V2

### How to reproduce

Not sure... Went to edit a profile as my Supports were weird...
-MY- edited version of the Profile has a setting for Support SPEED
-FACTORY- The profile does not have a speed setting.

Something is turning on/off the visibility of this setting, and I would like to know what it is so I can adjust it The 

### Actual results

Can't edit Support Speed as the setting does not appear

### Expected results

As I have selected ADVANCED all setting should show up

### Project file & Debug log uploads

[HeroMeBasics.zip](https://github.com/SoftFever/OrcaSlicer/files/15349960/HeroMeBasics.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]

igiannakas, 2024-05-19:
if you enable supports you'll see the support speeds.

inspectionsbybob, 2024-05-19:
Too Damm easy...
Need more Coffee

closed 2024-08-25 by github-actions[bot]