OrcaSlicer/OrcaSlicer #5389: octoprint device does not remember username/password for web login

nilesjohnson, 2024-05-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

pop os 22.04

### Printer

Creality Ender 5 Plus

### How to reproduce

1. Follow instructions to connect with octoprint using api key
2. in Device tab of Orca Slicer, enter username/password for octoprint web interface and check the "Remember Me" box
3. restart Orca Slicer
4. Device tab again requires username/password for octoprint web interface

### Actual results

Orca Slicer does not remember the username/password

[image]

### Expected results

I would like Orca Slicer to store the user/password for octoprint, in the same way that my web browser does.

### Project file & Debug log uploads

[debug_Fri_May_17_13_4.zip](https://github.com/SoftFever/OrcaSlicer/files/15355339/debug_Fri_May_17_13_4.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

I'm using the latest octoprint version: 1.10.1.
The API key connection from Orca Slicer to octoprint is working fine, even after restarting both: the "Print" button can successfully upload gcode and start a print.

Maybe relevant: I am using two different computers over a local network.
- basement laptop is wired to the printer and runs the octoprint server
- upstairs desktop is where I run Orca Slicer and use web browser to connect to octoprint

didli, 2024-05-17:
I second this (Fedora 39 with orcaslicer appimage). Setup is basically the same as @nilesjohnson.

ericthelin, 2024-06-20:
+1 I came here just to add this issue

otto2704, 2024-08-07:
+1 same here (orca newbie)

jeremiahputnam, 2024-08-08:
Same here, even with an API key in Orca Slicer.

UltraSalem, 2024-08-31:
same issue here. makes the Device pane kind of pointless as it's easier to just switch back to my browser each time I want to interact with octoprint.

otto2704, 2024-12-01:
still no progress :-(

xtreye, 2024-12-23:
Same issue with 2.2.0 under OpenSuse Leap. Also, sometimes the updates just freeze on the tab and I cannot reload one like I would do in a web browser.

jokesters, 2025-01-22:
> Same issue with 2.2.0 under OpenSuse Leap. Also, sometimes the updates just freeze on the tab and I cannot reload one like I would do in a web browser.

if you right click a menu will pop up by the mouse cursor and has an option to reload the page. But I also had this problem until I downloaded the autologin plugin and now as long as its on the same network it just logs into octoprint automaticlly..

xtreye, 2025-01-25:
> > Same issue with 2.2.0 under OpenSuse Leap. Also, sometimes the updates just freeze on the tab and I cannot reload one like I would do in a web browser.
This helps, thanks
 
> if you right click a menu will pop up by the mouse cursor and has an option to reload the page. But I also had this problem until I downloaded the autologin plugin and now as long as its on the same network it just logs into octoprint automaticlly..
A little bit of a problem here that this is a security hole ;)

skandalfo, 2025-04-02:
The development version of wxWidgets (3.3.0) seems to have introduced partial support for persistent cookie storage for web views:

https://docs.wxwidgets.org/latest/classwx_web_view_configuration.html#ae0f9c2280b50600e477dd2567ea992bb

closed 2025-07-09 by github-actions[bot]

saenglert, 2025-09-22:
Still same behavior in 2.3.1-beta

SoraMakes, 2025-12-01:
Seems to be a Linux only issue. Did not have this issue on Windows. I am using cachyos
Btw: still an issue in 2.3.1

Cobwebblox, 2025-12-23:
A good ol +1 here
Orcaslicer 2.3.1 (f2d10a0) (app image)
Bazzite (fedora based)

Print button works, device tab needs login every time