OrcaSlicer/OrcaSlicer #5391: Crash when using cut and trying to pan

jon-daemon, 2024-05-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

Mac OS 11.7.10

### Printer

Creality Ender 3 V3 S3

### How to reproduce

I drag a model on the plate.
Select the cut tool to cut it in half.
Use right click on mouse to move the plate.
It's not happening with every model.

### Actual results

The app is crashing.

### Expected results

Move the plate around

### Project file & Debug log uploads

[hollow_cube.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/15362525/hollow_cube.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

sdbruder, 2024-07-12:
for me that's a guaranteed crash to the desktop: load model, cut, right click. currently on v2.1.1.

jon-daemon, 2024-10-12:
It's still an issue for me on 2.1.1 macOS 13.7

jon-daemon, 2025-01-12:
It's still an issue

karlingen, 2025-03-06:
+1. Crashing here too