OrcaSlicer/OrcaSlicer #5395: replace accel to deaccel option with min cruise ratio

Nathan22211, 2024-05-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

klipper recently replaced accel to deacel with min cruise ratio, the currently used arg in orca gcode is deprecated

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

replace the aceel to deaccel with min cruise ratio

### Describe alternatives you've considered

this is because of the deprecation of a feature in Klipper, so there are no alternatives

vgdh, 2024-05-19:
the owner about this issue https://github.com/SoftFever/OrcaSlicer/issues/4521#issuecomment-2002295479

vgdh, 2024-05-19:
similar https://github.com/SoftFever/OrcaSlicer/issues/4521#issuecomment-2002295479

Tarakesh83, 2024-05-19:
as a quick fix:
just disable accel_to_decel in orca, enter the desired ratio in your klipper config and done.
and ignore this option until its removed in a future release.

closed 2024-08-25 by github-actions[bot]