OrcaSlicer/OrcaSlicer #5404: Is it possible to have build plates bigger than 2147 x 2147?

PolGuixe, 2024-05-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I can't add printers with bigger build plates than this.

Is this limitation related to the use of a 32 bit integer? 

See: https://github.com/SoftFever/OrcaSlicer/blob/dd36d5b1eaba4b7c35d392dadcfc686a37e20cf5/src/libslic3r/libslic3r.h#L57C3-L57C4

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

In the printer config I should be able to configure bigger build plates than 2147 x 2147.

### Describe alternatives you've considered

NA

buzzhuzz, 2024-05-20:
Implemented per https://github.com/SoftFever/OrcaSlicer/pull/5320

Try daily build version.

PolGuixe, 2024-05-20:
@buzzhuzz perfect!

tlhintoq, 2024-05-24:
I just gotta ask...

What printer has a bed larger than 2m square?

PolGuixe, 2024-05-27:
@tlhintoq concrete 3D printers for houses 😅

PolGuixe, 2024-05-27:
Solved in https://github.com/SoftFever/OrcaSlicer/pull/5320

closed 2024-05-27 by PolGuixe