OrcaSlicer/OrcaSlicer #5410: Can't add custom camera source in nightly on macOS

Starwalker98, 2024-05-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 nightly

### Operating System (OS)

macOS

### OS Version

macOS 14.4.1

### Additional system information

mac Mini M1

### Printer

Bambu Lab A1 Mini

### How to reproduce

Go to Devices
Click on Camera settings
Try to input anything into the "Hostname or IP" field

### Actual results

Whenever trying to type anything into the filed, the popup window goes away

### Expected results

Should be able to input manual camera source

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/15380939/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

This exact same thing and setup works on Windows

jjleto, 2024-06-21:
Same situation here 

OrcaSlicer Version
2.1.0-beta

Additional system information
Mac Mini M1

Printer
Bambu Lab A1 Mini
X1C

Starwalker98, 2024-08-11:
Still doesn't work with 2.2.0-dev.

stuomas, 2024-08-13:
Hi, did the feature ever work on MacOS? It was originally tested only on Windows, because the tester (=me) didn't have access to a Mac

stuomas, 2024-08-13:
Meanwhile, could you try to manually add to `OrcaSlicer.conf` your camera, for example
```
"camera": {
    "custom_source": "http://commondatastorage.googleapis.com/gtv-videos-bucket/sample/BigBuckBunny.mp4",
    "enable_custom_source": true
},
```
anywhere in the configuration JSON.

jjleto, 2024-08-13:
> Meanwhile, could you try to manually add your camera to `OrcaSlicer.conf`, by creating a `camera` object for example
> 
> ```
> "camera": {
>     "custom_source": "http://commondatastorage.googleapis.com/gtv-videos-bucket/sample/BigBuckBunny.mp4",
>     "enable_custom_source": true
> }
> ```
> 
> anywhere in the configuration JSON, while OrcaSlicer is NOT running. And then start OrcaSlicer.
> 
> This is all that the popup window does, so it should work this way unless there is some more Mac incompatibilities. I suggest trying with the BigBuckBunny link first to rule out any issues with the camera address

I tried but starting OrcaSlicer all "camera object" is instantly deleted from OrcaSlicer.conf.

Starwalker98, 2024-08-13:
Hi! No worries, thanks for getting to it! 
I do not know if it ever worked on Mac, I only got a 3D printer in April I think, but I wasn't able to have it work ever.

I have added the camera object, and it works. I can click Enable / Disable here, and it switches between the A1 mini camera and the Big Bunny video:
[image]

However, when I click on the the address bar to change it, the popup window goes away instantly, does not let me change it, but the bunny video continues to play.

So I then tried to change it in the config file, adding my RTSP source in the config file worked with the video working in slicer.

So I guess changing it / adding it in the config file works, but in the GUI it does not.

If there's anything else I can help with, let me know.

stuomas, 2024-08-13:
Ok, thanks for testing!

> adding my RTSP source in the config file worked with the video working in slicer
I'm not sure what you mean, but RTSP should not work currently directly, it needs URL to MJPEG stream or mp4 etc.

> I tried but starting OrcaSlicer all "camera object" is instantly deleted from OrcaSlicer.conf
You probably put it somewhere wrong that caused syntax error in the JSON

Starwalker98, 2024-08-13:
Ah, you are correct, it was a URL to MJPEG stream, my bad!

jjleto, 2024-08-14:
Ok it's working 

edit OrcaSlicer.conf

insert:

    "camera": {
        "custom_source": "http://youraddress/yourstream.html",
        "enable_custom_source": true
    },

don't forget the comma after the last bracket "   },"

I use frigate nvr which uses go2rtc to handle rtsp streams so I did the conversion to http with this.

Starwalker98, 2024-08-14:
> @Starwalker98 I made one blind attempt to fix it. If you want to test it, you can download OrcaSlicer from my fork here https://github.com/stuomas/OrcaSlicer/actions/runs/10386235291 (I assume `OrcaSlicer_Mac_arm64_V2.2.0-dev` in your case)
> 
> If it works, I'll make a PR

Hi! I've tried it, but it still goes away as soon as I try to type anything in the address bar.

stuomas, 2024-08-14:
> Hi! I've tried it, but it still goes away as soon as I try to type anything in the address bar.

Ok :( just to make sure, does it go away when you click it, or after you've tried to type first character?

Starwalker98, 2024-08-14:
Sorry! If I open the camera setting for the first time since launching Orca, it goes away after I click into the address bar. However, if I then open the setting menu again, I can click into the address bar, I can select text with my mouse, however any keyboard input makes it go away, both letters and arrows.

Perchycs, 2024-11-07:
A slight workaround I found is that you can copy the link, right click the text field rather than typing or cmd-v, and then click paste.

EnygmaINC, 2025-01-23:
Same issue. Latest MacOS, Macbook Pro (M3 Pro)

closed 2025-05-01 by github-actions[bot]