OrcaSlicer/OrcaSlicer #5417: Question about wall line width

olcayoren, 2024-05-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Why does the filament coming out of the 0.4mm nozzle appear to be 0.45? What is the mathematical formula for this?
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

None

buzzhuzz, 2024-05-21:
https://manual.slic3r.org/advanced/flow-math

olcayoren, 2024-05-22:
> https://manual.slic3r.org/advanced/flow-math
> 
> Just realized that question not about math itself, but regarding settings handling.
> 
> In your case better to express width in percentage: set it 100% to explicitly make it to be deducted from the nozzle size.
> 
> Seems like setting it to 0 exposes some bug in configuration.

Yes, your determination is correct, when you set the default width to 0, the width is 0.45mm. But when I set it to 100% it shows 0.40mm.

@SoftFever Is this a bug?

SoftFever, 2024-05-23:
0 means automatic or default in this case. When you put a absolute non zero value or percentage, Orca will calculate it accordingly

closed 2024-05-26 by olcayoren