OrcaSlicer/OrcaSlicer #5421: Allow changing infill combination layer height

Ataraxia-Mechanica, 2024-05-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently infill combination height seems to be tied to nozzle width, e.g. always 0.4mm when using a 0.4mm nozzle. This makes some 3d infills unstable especially at low infill percentage. I want to combine infill every 2 layers when printing at 0.1mm layer height.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add a custom config option that changes how many layers got combined/combined infill layer height.

Ataraxia-Mechanica, 2024-05-22:
Or just implement #2735 which proposes to make infill combination layer height limited by max layer height.

Ataraxia-Mechanica, 2024-05-22:
I don't have examples of worsening infills right now but this post illustrates what I meant: 
[https://www.reddit.com/r/FixMyPrint/comments/1cxdgmw/](https://www.reddit.com/r/FixMyPrint/comments/1cxdgmw/)
Basically not enough layer adhesion between infill causing gyroid to form this grid-like pattern.

Gamerou, 2024-05-22:
I cannot use infill combination in my prints because of this. I have max layer height set to 0.2mm on my 0.4mm nozzle and infill combination still wants to print 0.4mm high infills.

Gamerou, 2024-05-22:
same as https://github.com/SoftFever/OrcaSlicer/issues/2175

baros008, 2024-07-08:
Yes, please implement this setting.

idkjustgoogleit, 2024-07-12:
+1

pleaaaasseeee!?

ZedOud, 2024-07-26:
Also related but not yet mentioned: #4066 #2757

ZedOud, 2024-07-26:
It well known that layer adhesion really only works up until around layer heights of 80% of the nozzle width, so when printing at the common layer height of 0.2mm on a 0.4mm nozzle, infill combination layers are 0.4mm, which is impractical.

Also, in the case of printing at less than 33% of nozzle width, like 0.1mm layers on a 0.4mm nozzle, layer combination still results in results in 0.4mm layer heights, when 0.3 or 0.2 infill combination layer heights might be desirable.

Printing at about 3/8 or about 40% layer height (0.15-0.16mm on a 0.4mm nozzle producing 0.3-0.32mm combination layers) ends up being the only usable layer height. (3/8 to 40% also happens to be, coincidentally, the [optimal layer height for maximum layer adhesion](https://www.cnckitchen.com/blog/the-influence-of-layer-height-on-the-strength-of-fdm-3d-prints).)

Gamerou, 2024-07-27:
To summarize:

It's well known that layer adhesion is effective only up to about 80% of the nozzle width. So, when printing at a common layer height of 0.2mm on a 0.4mm nozzle, having infill combination layers at 0.4mm is impractical.

Furthermore, if you print with layers less than 33% of the nozzle width, like 0.1mm layers on a 0.4mm nozzle, the combination still results in 0.4mm layers. This is not always desirable, as one doesn't always want to print with such fine layers. In these cases, 0.3mm or 0.2mm infill combination layer heights would be preferable.

The optimal layer height for maximum adhesion, which happens to be around 3/8 or 40% of the nozzle width (0.15-0.16mm on a 0.4mm nozzle, producing 0.3-0.32mm combination layers), is often the only usable layer height for practical infill layer combination.

But I think for most people is the speed aspect crucial here. With a normal layer height of 0.2mm, combining infill layers to 0.3mm or 0.32mm would allow for faster prints. It's not about combining entire layers, but that's beside the point. Every other slicer has this option. Why doesn’t Orca let the user decide?

idkjustgoogleit, 2024-07-27:
Maybe if orcaslicer would give an notification when the max flow rate is being exceeded?

That way you can decide per layer in the slicer if you should decrease speed or line width to decide what is faster. The legenda colours and determining exact details is not good enough to dive into details.

However, atm i think the case might be of what is better; faster printing and more risks on the print and hardware... Or less printing moves and safer overall? So speed is perhaps not the main goal of infill combination?

Gamerou, 2024-07-27:
> Maybe if orcaslicer would give an notification when the max flow rate is being exceeded?
> 
> That way you can decide per layer in the slicer if you should decrease speed or line width to decide what is faster or better. The legenda colours and determining exact details is not good enough to dive into details.
> 
> However, atm i think the case might be of what is better; faster printing and more risks on the print and hardware... Or less printing moves and safer overall? So speed is perhaps not the main goal of infill combination?

Good point. Ultimately, however, the user should decide for themselves whether they would prefer to print faster, which entails more risks for the print quality and the hardware, or whether they opt for fewer printing movements and an overall safer option.

igiannakas, 2024-08-09:
Raised a PR that aims to address this above.

closed 2024-09-07 by SoftFever