OrcaSlicer/OrcaSlicer #5430: libwebkit2gtk problems even after installing libwebkit2gtk-4.1-0

MFernflower, 2024-05-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 and the nightly build aswell

### Operating System (OS)

Linux

### OS Version

Xubuntu 24.04 LTS Noble Numbat

### Additional system information

core i5 system with 32 gb of ram running Xubuntu 24.04 LTS 

### Printer

CREALITY K1

### How to reproduce

Appimage fails to launch with error:

/tmp/.mount_OrcaSlsTuRmG/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Actual results

Appimage fails to launch with error:

/tmp/.mount_OrcaSlsTuRmG/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Expected results

I tried installing the latest libwebkit2gtk for my OS version but did not help at all

### Project file & Debug log uploads

Appimage fails to launch with error:

/tmp/.mount_OrcaSlsTuRmG/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Appimage fails to launch with error:

/tmp/.mount_OrcaSlsTuRmG/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

buzzhuzz, 2024-05-23:
Duplicate of #4616 

Attempt to fix it made in #5203, but there are issue found during code review which I don't actually know how to fix yet.

Luke56784, 2024-05-23:
You have to install 4.0. Not 4.1
But it's not in the repository's anymore with 24.04.

You also can link it by executing 

`sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
sudo ln -sf /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18`

 https://github.com/bambulab/BambuStudio/issues/3973

MFernflower, 2024-05-23:
> bambulab/BambuStudio#3973

I will try this when I get back home

Thank you

GA3Dtech, 2024-05-25:
I had same issue on Ubutu LTS 24.04 fresh install

I can confirm that linking path is working

thk

mikeperalta1, 2024-05-27:
+1 issue began immediately after upgrading to Ubuntu 24.04

ffries, 2024-06-20:
All these issue are related :
https://github.com/SoftFever/OrcaSlicer/issues?q=libwebkit2gtk-4.0.so.37

MFernflower, 2024-06-20:
@ffries the two bash commands given by Luke fix the issue

closed 2024-07-03 by MFernflower