OrcaSlicer/OrcaSlicer #5431: Dynamic (calculated) line widths not treated correctly by other fields.

tlhintoq, 2024-05-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1 nightly 22may2024

### Operating System (OS)

Windows

### OS Version

Win10x64pro

### Additional system information

Core i9 11th gen
64gig ram
Nvidia

### Printer

Any/all

### How to reproduce

Set nozzle diameter - 0.6 in this case
Set a dynamic default line width so the same profile works on a .4 machine as well as a .6 machine.  In this example `150%`.
Set other lines widths to `0` so they automatically inherit the line width from `Default line width`.  In this example that makes them all `0.8`
Tree supports: Set tip diameter to 0.8mm

### Actual results

All the math works out that everything is 0.8 in this example.  Yet you can't slice this because of error 
`Organic support tree tip diameter must not be smaller than support material extrusion width.`

NOTE:  Calculated line widths are also broken if you paint the object with different materials/colors.  Basically the calculated width feature which would be awesome is dramatically crippled to the point of not being usable by other elements/features of Orca.

### Expected results

The flow of the math, the order of operations carried out, should calculate that all the inherited line widths are 0.8 and therefore this error should not arise.

### Project file & Debug log uploads

No project file required as this is reproducable from the instructions above.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]

tlhintoq, 2024-05-23:
And just to confirm its a math problem... 
Change the `150%` to `0.8` and the error goes away - even though the result should be the same.
Just a guess... loss of precesion in data types?  `Float` to `decimal` or something like that causing ` decimal 0.6` times `float 150%` to come out `0.79999999999` or something similar

tlhintoq, 2024-06-19:
Latest 2.1 RC still doesn't address the use of % line widths.  Its such a great option that I would love to be able to use it.  
Its especially nice when you have machines with different nozzle sizes installed.  Great to just set a profile with `150%` for example.
But you can't use these dynamic values if you do anything beyond the ultra basics of Orca.
Paint the object - Breaks
Dynamic layer height - breaks

[image]

[image]

CC: @SoftFever

SoftFever, 2024-06-20:
@tlhintoq Can you attach a sample project?

tlhintoq, 2024-06-21:
[image]

[image]

[DoubleBinV5.zip](https://github.com/user-attachments/files/15921674/DoubleBinV5.zip)

tlhintoq, 2024-06-21:
Wait... It may not be the use of % but instead the computed value.
If I hard code the top layer line width to .24 I get the same error.
I can go as low as .28 and its acceptable and slices.

tlhintoq, 2024-06-21:
If you get a chance... Can you shed some light on `Flow::with_spacing()` ?
I'd like to have more of a true understanding of where this is coming in to play so I can be smarter about some of problems.
Is there some `flow spacing` or other `spacing` value this is interacting with and that's where the conflict arises?  Or just a hard value that says: "Can't go smaller than 70% of the nozzle width"... something like that?