OrcaSlicer/OrcaSlicer #5438: After Upgrade - Requires The Application To Be Opened Twice Before User Profiles For Printers And Filament Are Available

arch3angel, 2024-05-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Microsoft Windows 10 Pro - v10.0.19045 Build 19045

### Additional system information

System Type: x64-based PC
Processor: Intel(R) Core(TM) i7-10875H CPU @ 2.30GHz, 2304 Mhz, 8 Core(s), 16 Logical Processor(s)
Physical Memory (RAM): 64.0 GB
Graphics: NVIDIA GeForce RTX 2070 Super With Max-Q (8 GB GDDR6)

### Printer

Creality K1 Max - But Happens With All 18 Printers I have (Mix Match Of Printers)

### How to reproduce

1. Go to start menu, click the OrcaSlicer icon to start the application
2. OcraSlicer opens and when you go to the drop down for the printers, no User Presets are visible.
3. Close the application
4. Reopen the application
5. when you go to the drop down for the printers, everything visible.

This is not an issue with a project, it happens even when trying to start a new project.

### Actual results

All User Presets are visible

### Expected results

All User Presets become visible

### Project file & Debug log uploads

This is not an issue with a project, it happens even when trying to start a new project.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

N/A

winbard, 2024-05-25:
Hi,

I have the same problem. Orca Slicer 2 with Windows 10 Pro. I unistall Orca, delete all files on pc, clear regedit.
New install the same problem.
After the instalation i chose one printer (K1max) and after that i load my profiles from orca slicer back.
In the version from orca slicer before i had not this problem.

epicylon, 2024-06-01:
I also have the same problem on 2.1.0-beta

mindkill666, 2024-06-02:
Same here, i went back to version 1.9.1. 
No problems with that version.

sam0622, 2024-06-07:
I have the same issue on 2.0.0 and 2.1.0-beta

VANHEKENT, 2024-06-25:
As do i. I have had the issue since the last two official updates.

closed 2024-10-01 by github-actions[bot]