OrcaSlicer/OrcaSlicer #5449: Flatpak Build Does Not Render X1 Camera

rehashedsalt, 2024-05-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-dev

### Operating System (OS)

Linux

### OS Version

Fedora 40 Atomic Desktop, Flatpak

### Printer

Bambu X1CC

### How to reproduce

1. Install OrcaSlicer from Flatpak
2. Connect to Bambu printer
3. Hit "play" on camera feed

### Actual results

Camera feed yields the following error:

[image]

### Expected results

Camera feed should render the interior of the printer

### Project file & Debug log uploads

Reproducible from fresh project.

Debug logs are a little sparse --

```
[warning]       2024-05-21 13:25:57.121075[Thread 0x00007f30d4fbab80]:get_network_function, can not find function bambu_network_del_subscribe
[warning]       2024-05-21 13:25:57.121175[Thread 0x00007f30d4fbab80]:get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]       2024-05-21 13:25:57.121191[Thread 0x00007f30d4fbab80]:get_network_function, can not find function bambu_network_get_profile_3mf
[error] 2024-05-21 13:25:57.451196[Thread 0x00007f3037e006c0]:DBus Request name Error: org.freedesktop.DBus.Error.ServiceUnknown
[error] 2024-05-21 13:25:57.451210[Thread 0x00007f3037e006c0]:Dbus Messages listening terminating.
[warning]       2024-05-21 13:25:59.456574[Thread 0x00007f30d4fbab80]:01:25:59 PM: Warning: Error running JavaScript: Unsupported result type
[warning]       2024-05-21 13:25:59.480879[Thread 0x00007f30d4fbab80]:01:25:59 PM: Warning: Error running JavaScript: Unsupported result type
[warning]       2024-05-21 13:25:59.483481[Thread 0x00007f30d4fbab80]:01:25:59 PM: Warning: Error running JavaScript: Unsupported result type
[warning]       2024-05-21 13:25:59.484040[Thread 0x00007f30d4fbab80]:01:25:59 PM: Warning: Error running JavaScript: Unsupported result type
[warning]       2024-05-21 13:25:59.891446[Thread 0x00007f17ff4006c0]:[Orca Updater]: online version=1.9.1.6, current_version=1.9.1.6, no need to download
[error] 2024-05-21 13:26:09.704801[Thread 0x00007f30d4fbab80]:send_message_to_printer error: network_agent=0x55c37f07ae30, ret=-4, dev_id=00M00A2A1404827, json_str={"info":{"command":"get_version","sequence_id":"20000"}}, qos=1
```

-- so the program was run with `GST_DEBUG=4`, which adds tons of debug logging information from gstreamer. Here's Orca's STDOUT/ERR log in that case:

[orca.log](https://github.com/SoftFever/OrcaSlicer/files/15441024/orca.log)

To compare against, Bambu Studio's Flatpak does not have this issue. Here is a log from Bambu Studio (1.9.0.70) with the same gstreamer debug level:

[bambu.log](https://github.com/SoftFever/OrcaSlicer/files/15441026/bambu.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Spun off from PR #4175, cc @Alaric323 who also had the issue and @powpingdone, who did a stellar job with the Flatpak!

First thing that comes to mind is some random dependency incongruity between the two packages, probably related to gstreamer or a plugin thereof.

Alaric323, 2024-05-25:
As an aside, I had the issue on OpenSUSE tumbleweed.  I don't know if that has much effect on a Flatpak though since they usually have their depends bundled.

I'm away from my computer for the next week though so I can't test it on OpenSUSE.  Can check Ubuntu though on a device I have with me.

image.png (view on web)<https://github.com/SoftFever/OrcaSlicer/assets/12635110/5b32e1c4-341f-4d7c-a3b8-03e88bde22a2>

  *    Log file

powpingdone, 2024-05-25:
It's defo a dependency issue regarding gstreamer, though it could also be that some h264 codec isn't being requested by the Flatpak package (patent issues, you know the drill). It can be done easily, but I don't know if that's specifically the issue.

powpingdone, 2024-05-25:
Hmmm. https://github.com/bambulab/BambuStudio/issues/3279

Alaric323, 2024-05-28:
Same issue occurs on Ubuntu 24.04, so it's definitely a bundled dependency thing.  The log files found at ~/.var/app/io.github.softfever.OrcaSlicer/config/OrcaSlicer/log are basically useless by the looks of things, the data in it appears to only be related to printer/filament profiles.

Does make me curious as to whether or not a self-hosted filament storage website could be made to clone BambuLab's implementation though...

powpingdone, 2024-08-27:
@Alaric323 @rehashedsalt Does the artifact here fix the issue? https://github.com/SoftFever/OrcaSlicer/actions/runs/10570248432/artifacts/1857743577

rehashedsalt, 2024-08-27:
That build works on my machine! Camera starts up and renders fine when I hit play, doesn't seem to have any issues.

Alaric323, 2024-08-28:
I can also confirm it works on an up to date OpenSUSE Tumbleweed install using Wayland.  A little grainy but I don't know if that's my wifi or the slicer at the moment.

anarsoul, 2024-11-22:
Please try if https://github.com/anarsoul/io.github.softfever.OrcaSlicer/releases/tag/v2.2.0-2 fixes the issue. It has a patch from @powpingdone

Wiggoggs, 2025-01-19:
> Please try if https://github.com/anarsoul/io.github.softfever.OrcaSlicer/releases/tag/v2.2.0-2 fixes the issue. It has a patch from [@powpingdone](https://github.com/powpingdone)

Found this thread while searching for a solution to the same problem, can confirm that this flatpak v2.2.0-2 works on bazzite

closed 2025-02-17 by SoftFever