OrcaSlicer/OrcaSlicer #5455: plate and file names

quadlike, 2024-05-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2

### Operating System (OS)

Windows

### OS Version

win10

### Printer

P1S and P1P

### How to reproduce

1. add few files to project
2. place models in different plates
3. rename plates
4. tap on "print"
5. look to file name

### Actual results

The name of the plate is not transferred to the name of the print file. If there are many different models in the project. It's very inconvenient. If there is such a possibility, please fix it

### Expected results

It is necessary that the name of the plate be transferred to the file name. I have 18 printers, and I need to write the file name every time I send it. and I have projects with 20 tables. It
takes a very long time.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2024-05-29:
has been fixed here: https://github.com/SoftFever/OrcaSlicer/pull/5401

quadlike, 2024-05-30:
Great! thx a lot!

closed 2024-09-05 by github-actions[bot]