OrcaSlicer/OrcaSlicer #5465: Scarf joint seams go below set flow rate (and speed)

dburburan, 2024-05-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

GCode preview

### How to reproduce

Set a low flow rate restriction like 3mm³/s in the filament properties.
Since this causes print speeds well below what the printhead is capable of - the slicer should be able to maintain exactly this flowrate everywhere by adjusting the printhead speed. Currently this is true on all features except scarf joint seams.

### Actual results

3mm³/s was set here - but you can see on the scarf the flow rate drops well below that.
[image]

Note that this is different from the normal case where the printhead is using the wall speed setting - there it's necessary to drop the flow rate for the scarf as the print head is already speed restricted.

However here - for the duration of the scarf - the printhead is both travelling below the set speed and below the set flow rate.

### Expected results

It looks like the volumetric flow limiter is being calculated first and then the scarf speed is calculated relative to that. But I think it should be the other way around, where the flow limit is imposed as the final step.

The end result is I would expect the flow rate to be maintained while the printhead accelerates through the scarf - in order to produce the same effect.

Is my understanding correct? It could be important e.g. when printing TPU or with bowden tubes as problems are more likely when flow rates fluctuate.

### Project file & Debug log uploads

I just used a default cylinder primitive

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DaveMeade5, 2024-08-18:
I have also had this problem. The scarf joint pushes the volumetric flow almost down to zero, causing very rough outer layers. 
These all have have scarf joint enabled
[image]
[image]
[image]

This is the texture it produces 
[image]
[image]

This is no scarf joint enabled
[image]

Its way worse with 'scarf joint across entire wall' enabled

boromyr, 2024-09-26:
I also have the same problem, and the scarf is completely unusable, does anyone know how to make it work?

en-vy, 2024-11-22:
Same issue here.

vmakeev, 2024-12-08:
Unfortunately, I encountered a similar problem

[image]

closed 2025-03-17 by github-actions[bot]