OrcaSlicer/OrcaSlicer #5474: Graphical errors with Flatpak

cochcoder, 2024-05-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-alpha

### Operating System (OS)

Linux

### OS Version

Fedora 40 (KDE 6.0.5)

### Additional system information

CPU: Intel i5-10210U
GPU: integrated graphics
Memory: 8 GB

### Printer

Ender-3

### How to reproduce

1) Install Flatpak
2) Go to prepare tab
3) Move your mouse

### Actual results

The group of tools will flicker until you close your printer's menu, the cube to move your camera has a different background color than the application, and the flatpak fails to switch to dark mode when the desktop environment is in dark mode. 

### Expected results

The tool group shouldn't flicker, the cube to move the camera should have the same background color as everything else, and the flatpak should be in dark mode when the desktop environment is in dark mode. 

### Project file & Debug log uploads

[Project File](https://github.com/SoftFever/OrcaSlicer/files/15449557/bug.zip)
[log](https://github.com/SoftFever/OrcaSlicer/files/15449566/debug_Sun_May_26_20_29_45_4.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

@powpingdone I'm not sure how much of this is the flatpak or just Orca Slicer as I was only able to test the Appimage two weeks ago and I am unable to currently test it. 

[image]

[Screencast_20240526_201838.webm](https://github.com/SoftFever/OrcaSlicer/assets/103969142/2d67c0b0-0012-44ae-a6cf-f1b876399613)

Alaric323, 2024-05-28:
This occurs with me as well, but only when the window is smaller than the top toolbar.  As shown in cochcoder's video, the graphical glitchiness stops when the sidebar is hidden and the full top toolbar is rendered.

powpingdone, 2024-05-29:
Sorry sorry, following this now. When work stops being so intense I'll look into this.

cochcoder, 2024-05-29:
You're good, thanks for bring this slicer to flatpak.

powpingdone, 2024-07-25:
@cochcoder is the graphical error still happening?

cochcoder, 2024-07-29:
@powpingdone The graphical error is still happening on my end and I still can't test to see if the same issue happens with Appimages.

cochcoder, 2024-10-28:
I tested the flatpak provided by @anarsoul in issue #3949 and no longer experience these issues.

closed 2024-10-28 by cochcoder

anarsoul, 2024-10-28:
FWIW I had this issue with appimage as well. It happens if the icons don't have enough space. It's not really related to flatpak.

BTW, it looks like you don't have Adwaita dark theme installed in your flatpak environment. `flatpak --user install org.gtk.Gtk3theme.Adwaita-dark` should fix it.