OrcaSlicer/OrcaSlicer #5478: "Raft contact Z distance" missing in Process

Ergonomicmike, 2024-05-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-alpha

### Operating System (OS)

Windows

### OS Version

Win10

### Printer

Ender 3 v2

### How to reproduce

In Orca,
Process - Global - Advanced
Search for "Raft." Among the entries will be "Raft contact Z distance." 
Click on it.

### Actual results

Nothing happens. It doesn't take you to a "Raft contact Z distance" setting. And there is no "Raft contact Z distance" setting in the Raft section under Support.

### Expected results

I expect to be able to go "Raft contact Z distance" and set a distance.

If Raft distance has been combined with some other support distance, then either 1) the Search function shouldn't return "Raft contact Z distance," or 2) the Search function should take you to the pertinent support setting.

### Project file & Debug log uploads

[ziptie_150mm_for_Orca_report.zip](https://github.com/SoftFever/OrcaSlicer/files/15455916/ziptie_150mm_for_Orca_report.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

buzzhuzz, 2024-05-27:
Set number of raft layers to non-zero value and you'll see "Raft contact z distance" config entry.

Ergonomicmike, 2024-05-27:
It's set to "1" now. I tried increasing it (2, 3, 4) before reporting the issue. No change here.

Ergonomicmike, 2024-05-27:
[image]

kisslorand, 2024-05-28:
This is what I have in Win11 with latest 2.1.0-alpha.

[image]

Ergonomicmike, 2024-05-28:
Tnx. I have the portable alpha from two days ago. (Latest, AFAIK.) I'll try again when the next update is released.

Ergonomicmike, 2024-05-28:
I just now downloaded the latest portable alpha. (5 hours old.) Still no Raft contact Z distance showing here.

Not even a placeholder for it. (So it doesn't appear to be a graphic card/driver problem on my end.)

Anyone able to reproduce this issue?

Ergonomicmike, 2024-05-29:
Just downloaded 2.1.0-beta. Still no Raft contact Z distance for me.

Can someone please load my project file and report what they see? Perhaps this issue is related to some other settings that I set in that project?

Ergonomicmike, 2024-06-04:
I just downloaded 2.1.0-beta_portable nightly, 7 hours old as of this writing. 

Raft contact Z distance works for a new project. (When raft layers > 0.) But still not working on the project that I uploaded here. (A zip-tie.)

So apparently there must be some setting in my project file that prevents/is incompatible with Raft contact Z distance. Or the model itself?

I'll try to experiment by toggling/trying other settings and report back.

Ergonomicmike, 2024-06-06:
Okay, I found what causes "Raft contact Z distance" to disappear.

I had set "Top Z distance" to 0 in my project.

"Raft contact Z distance" will show with any number > 0.

Should there be an advisory about this in the pop up for "Top Z distance"? 

I've edited the Title. I'll let the dev's close this "issue" after deciding its merits.

Ergonomicmike, 2024-06-26:
FWIW, choosing the (Experimental) Tree Slim style support sets "Top Z distance" to 0. 

But unchoosing it doesn't return Top Z distance to nominal.

closed 2024-10-03 by github-actions[bot]