OrcaSlicer/OrcaSlicer #5480: Be able to escape chars in filename format

SyFizz, 2024-05-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Hello,

I want to use [ and ] chars in my filenames, to output something like this :
[PLA] [210°C] Filename.gcode

But I get an error if i try to do so, even if I put escaping backslashes `\` before the brackets.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

When using backslashes or another escape char, the brackets should be escaped and not parsed as variables.

Ergonomicmike, 2024-05-27:
I am not disparaging the motive of the requestor and I can see how this Request could be helpful.

I merely offer that allowing escape sequences in input characters have been a vector for attacks - malware/hacks/heap overflows - in software/applications.

If allowed, there should be good input checking.

(I don't know why anyone would want to hijack Orca. But perhaps as a means to get control of an OS?)

SyFizz, 2024-05-29:
I see what you mean.
If not allowing escape sequences, maybe add an option to change the chars used for variables ? I mean, be able to change [variable] to {variable}

I'm sorry, my english is not good, it's not my native language.

closed 2024-09-04 by github-actions[bot]