OrcaSlicer/OrcaSlicer #5482: Unable to install Appimage on Ubuntu 22.04 - Fuse3 vs Fuse2?

FabianN, 2024-05-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04

### Printer

Bambulab P1S

### How to reproduce

1.  Add execution permissions to downloaded file "OrcaSlicer_Linux_V2.1.0-alpha.AppImage"
2. Run file "OrcaSlicer_Linux_V2.1.0-alpha.AppImage"

### Actual results

Get this output:
```
dlopen(): error loading libfuse.so.2

AppImages require FUSE to run. 
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information
```

### Expected results

install should run

### Project file & Debug log uploads

NA, can't install, no logs available

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I did some digging and it sounds like I need fuse2 for OrcaSlicer but it looks like I already have fuse3 installed and what I found suggests that downgrading my fuse version risks breaking other packages. Is there some way to get this to work with fuse3?

cochcoder, 2024-05-28:
Have you tried running `./OrcaSlicer_Linux_V2.1.0-alpha.AppImage --appimage-extract` in the folder it's downloaded?

Hjellvik, 2024-05-30:
> Have you tried running `./OrcaSlicer_Linux_V2.1.0-alpha.AppImage --appimage-extract` in the folder it's downloaded?

kristian@kristian-MAIN:~/Documents$ ./OrcaSlicer_Linux_V2.0.0.AppImage --AppImage-extract
/tmp/.mount_OrcaSlHqBRFg/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

Nathan22211, 2024-06-23:
the webkit2gtk issue is known to occur in modern distros as well. webkit2gtk and I think one other package need to be installed manually, especially on Arch distros

Samueru-sama, 2024-07-14:
The fuse error can be fixed by switching to this [appimagetool](https://github.com/probonopd/go-appimage) as it uses the static runtime.

closed 2024-10-20 by github-actions[bot]