OrcaSlicer/OrcaSlicer #5485: "other layers" bed temperature don't apply when disabling supports or disabling brims

SilencedFrost, 2024-05-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

11 Pro

### Additional system information

CPU: Ryzen 3 3300x
GPU: Asus Rog Strix 1070ti

### Printer

Neptune 4 Max

### How to reproduce

1. Go to filament settings, set first layer bed temperature to 70 degree, set other layers to 60 degree
2. In process, turn turn off supports/disable auto brim
3. Slice
4. Check gcode preview at the start of layer 2/end of layer 1

### Actual results

M104 is present (Nozzle temp), but M140 is not (Bed temp)
Printer does not lower temps after first layer because the gcode is not present

### Expected results

M140 S60 should be present
Printer should lower bed temperature to 60 after first layer

### Project file & Debug log uploads

[No bed temp change bug report.zip](https://github.com/SoftFever/OrcaSlicer/files/15467252/No.bed.temp.change.bug.report.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]

vgdh, 2024-05-28:
related
https://github.com/SoftFever/OrcaSlicer/issues/4676
https://github.com/SoftFever/OrcaSlicer/issues/3015

igiannakas, 2024-06-01:
@SoftFever 
I've come across this issue too today and I've been trying to fix this over the last couple of days but it's so difficult to reproduce. I currently have two Orca windows open - one run from Xcode and one compiled binary and the Xcode one just doesnt do it for some reason!

Smells like some race condition somewhere likely in the function below. It checks if the desired bed temperature is equal to the last bed temperature. However I notice that this variable is not initialised in the constructor, unless I'm missing something?

[image]

Constructor:
[image]

I'll try debugging this but it's almost impossible to replicate in my dev system vs. the operating binary. 

A workaround could be to disable the check and always emit the set bed temperature on each layer. That is not elegant but not the end of the world either? Assuming that the reason its not being set is that its failing that check, which I cannot prove :P

igiannakas, 2024-06-01:
PR pushed. Initialising the bed temperature variable with 0 in the constructor seems to have resolved this. Testers needed in the PR.

closed 2024-06-04 by SoftFever