OrcaSlicer/OrcaSlicer #5488: Embossed text and some other features disappear

Rickthebig, 2024-05-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

DIY CoreXY (RRF)

### How to reproduce

Slice the model using the provided project file and inspect the result

### Actual results

Embossed text (which technically are holes) mostly disappear. The preview shows strange "empty" layers in some parts of the model (for example, in the cylinder). When actually printed, the embossed text is hardly recognizable on the surface, which is closed instead of open. There are other unusual details, like a partially filled hole (the embossed star). The half-sphere is completely missing.

### Expected results

Holes should be empty. "Ghost" layers should not appear with marks showing seams where the layers do not exist (but should). 
This is the model (cutout):
[image]
This is what Orca slices:
[image]
This is what Cura slices and perfectly prints:
[image]

### Project file & Debug log uploads

[3D_test_V3.zip](https://github.com/SoftFever/OrcaSlicer/files/15477189/3D_test_V3.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Trying to get the model sliced correctly, following things were tried:
Changing layer thicknes from 0.2 to 0.08 mm
Changing wall generator Arachne <-> Classic
Changing Slice gap closing radius from 0.5 to .02
Changing resolution from 0.12 to 0.06
Changing X-Y hole compensation to 0.4: Some letters reappeared, but holes are too wide then.
And some other things. Nothing helped.

Avpman2, 2024-05-30:
Commenting so I can follow this thread. Having similar issues.

Gozania, 2024-06-01:
commenting due to a similar situation...  Orca text embossing giving undesired results, where as Prusa functions correctly.  A bonus feature/function that no other slicer has would be the ability to specify the amount of lines/walls for text.

Orca Embossing
[image]
[image]

Prusa Embossing
[image]
[image]

Hunted05, 2024-06-14:
Also have this issue.  Annoying as I noticed it half way through a multiple piece print.  Please fix

Gozania, 2024-06-14:
> Where is the "Only one wall on top surfaces" option located? tia

[image]

Avpman2, 2024-06-14:
I must be having a different issue. The option on my setup defaulted to unchecked.

RF47, 2026-10-07:
another issue:

[image]

This part has non-manifold edges, but they disappear when it is repaired.

closed 2026-10-13 by ianalexis