OrcaSlicer/OrcaSlicer #5489: Crash on startup MacOS - Error: OrcaSlicer got an unhandled exception: Invalid value provided for parameter filament_notes:

bdwilson, 2024-05-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

14.4

### Additional system information

M1

### Printer

Bambu Lab A1 Mini

### How to reproduce

Put this in the filament_notes field:
[warning]	2024-05-28 20:46:19.624391[Thread 0x0000000201acfac0]:08:46:19 PM: Error: OrcaSlicer got an unhandled exception: Invalid value provided for parameter filament_notes: "235/240C: 7mm3/s, K 0.125 Aux 70\n220/220C: 7.2 mm3/s K 0.26 Aux 20. Maybe max flowrate 6"\n\nFrom Here https://makerworld.com/en/models/476147?from=search#profileId-387036\n
[error]	2024-05-28 20:46:19.624399[Thread 0x0000000201acfac0]:Uncaught exception: Invalid value provided for parameter filament_notes: "235/240C: 7mm3/s, K 0.125 Aux 70\n220/220C: 7.2 mm3/s K 0.26 Aux 20. Maybe max flowrate 6"\n\nFrom Here https://makerworld.com/en/models/476147?from=search#profileId-387036\n

### Actual results

[warning]	2024-05-28 20:46:19.624391[Thread 0x0000000201acfac0]:08:46:19 PM: Error: OrcaSlicer got an unhandled exception: Invalid value provided for parameter filament_notes: "235/240C: 7mm3/s, K 0.125 Aux 70\n220/220C: 7.2 mm3/s K 0.26 Aux 20. Maybe max flowrate 6"\n\nFrom Here https://makerworld.com/en/models/476147?from=search#profileId-387036\n
[error]	2024-05-28 20:46:19.624399[Thread 0x0000000201acfac0]:Uncaught exception: Invalid value provided for parameter filament_notes: "235/240C: 7mm3/s, K 0.125 Aux 70\n220/220C: 7.2 mm3/s K 0.26 Aux 20. Maybe max flowrate 6"\n\nFrom Here https://makerworld.com/en/models/476147?from=search#profileId-387036\n

### Expected results

No error. 

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [ ] Project file

bdwilson, 2024-05-29:
Removed the URL from filament_notes and Orca started again.

SoftFever, 2024-05-29:
Looks like it's not reproducible in the nightly build.

[image]

closed 2024-09-04 by github-actions[bot]