OrcaSlicer/OrcaSlicer #5491: Broken K1/K1C/K1 Max Profile

robomopa, 2024-05-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Creality K1 Max

### How to reproduce

1. Set up Printer
2. Select any Creality K1, K1C, K1 Max Profile
3. Put an stl file with or without changing Printer, Material, Process settings
4. Slice
5. Go to Preview Screen
6. Select Temperature in dropdown menu

### Actual results

In dropdown menu print(nozzle) Temperature shows 0
Also Printer give error or if default set value is available tries to print with default bed and extruder temps, rather then the values selected in the slicer. (in k1 max 210degrees nozzle, 50 degrees bed)

Same steps with same stl with different printer profiles not give this issue. It happens only in Creality K1, K1C, K1Max profiles.

### Expected results

The temperature dropdown menu in slicer preview screen must show the set temperature in the material profile.
Printer must use the slicer temperature values.

### Project file & Debug log uploads

[orca_k1max_temperature.zip](https://github.com/SoftFever/OrcaSlicer/files/15479822/orca_k1max_temperature.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/15479832/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]

bistory, 2024-05-30:
It's not a bug in the profiles, it's a bug in Orca because it reads gcode commands and is able to use temperatures set in macros.
FYI setting 

```
M190 S[bed_temperature_initial_layer_single]
M109 S[nozzle_temperature_initial_layer]
```

Before calling the start_print macro solves the issue in Orca but creates problems on K1 printers.
So the best you can do is ignore that bug ;-)

robomopa, 2024-05-30:
In the 2.0.0 version printer also seems that it cant process the temp values. In the last beta, I tried without changing the gcode (not adding m190 code) again preview screen shows 0 but printer works as usual. Thanks for reenabling orca for me. I am closing this issue as solved. Pleqse fix the bug in official version. Thanks.

closed 2024-05-30 by robomopa

eccentricexit, 2024-05-31:
I am facing the same issue with my Anycubic Kobra Max.

I believe this should not be kept open since it is a bug in Orca that we would want solved.

sk8board, 2024-06-18:
@robomopa I agree with @eccentricexit. As mentioned previously, this is a bug. Therefore, could you reopen this issue as a bug?