OrcaSlicer/OrcaSlicer #5493: Variable Layer High does not work with cleaning tower when there are different Objects

Lummsky, 2024-05-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am using variable layer high on an multicolor Object and ad put other objects to the bed. 
Then Orca says: Cleaning Tower is not possible when the objects have different Layer high. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be nice if its possible to have a button like "put the varibale layer high to all objects"

16gadge, 2024-05-29:
yes that would help so much

Lummsky, 2024-06-03:
Perfect. Thank you a lot!
In the project I am using this function it woked perfectly fine.

closed 2024-09-09 by github-actions[bot]