OrcaSlicer/OrcaSlicer #5494: Stack Objects. Put different material to devide parts.

Lummsky, 2024-05-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I trie to stack parts(PLA-Tough). So in my case it´s possible to print at leat 100 parts during one night instead of just 30.

So I put the on top of each other and am using a third part to devide and support them. This part is printed with PETG. This makes it possible to have clean surfaces and seperate these parts easily. But When I stack these three(or more parts) to a tower Orca makes them to one part while printing and does not print top-layers to the end of Objects.

I do th
is with 100% infill settings put the printer does not slow down speed or fan for the new 1st layer withe new material and it does not fill gaps like on top or groundlayers.

[image]
[image]
[image]
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I want to have the option to stack parts and print them like if they where laying on the bed. 
Including first layer settings like speed and cooling and closing gaps etc.

There is enormous potential to make series parts on small printers.

[image]

rsglobal, 2024-08-16:
That works in most cases, even without support material. Just creating a gap between objects is enough. For PETG - 1 layer gap is good. For the ABS - you need less than that, like half a layer gap.

There's no issue to stack them together in slicer by grouping it into an assembly and using move command.

Usually, parts can be separated by bare hands using a little force.

However, it would be nice to be able to select a gap between parts in millimeters, not in layers, because you may want to find the sweet spot for your material and printing parameters.

closed 2024-11-22 by github-actions[bot]

GavinHanson, 2024-12-17:
Can this be re-opened? with the rise of tool changing printers this feature would be invaluable to be able to easily 'stack' STL's with a pre-defined support material.

Whilst yes its possible via workarounds its not ideal and is very time consuming.