OrcaSlicer/OrcaSlicer #5499: create printer - some vendors are not found

HAndreasH, 2024-05-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-alpha

### Operating System (OS)

Windows

### OS Version

Win 11

### Printer

Ender 3 pro

### How to reproduce

create printer/nozzle dialog
some vendors are not selectable
noticed it by selecting Creality

[image]

### Actual results

error message: The model was not found; please reselect vendor.

### Expected results

select model from vendor

### Project file & Debug log uploads

[debug_Wed_May_29_16_44_26_17932.log.zip](https://github.com/SoftFever/OrcaSlicer/files/15487640/debug_Wed_May_29_16_44_26_17932.log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

HAndreasH, 2024-05-29:
just tested in V2.1.0-beta, same result

bistory, 2024-05-30:
Same issue with Bambulab, Comgrow, "Custom Printer", Elegoo, Flashforge, FlyingBear, Folgertech, InfiMech, Kingroon, "Orca Arena Printer", Peopoly, Prusa, Qidi, Raise3D, SecKit, Tronxy, Twotrees, UltiMaker, Vivedino, Voxelab, Vzbot and Wanhao printers.
I guess there is something missing in the profiles.

HAndreasH, 2024-06-16:
still not working in V2.1.0-rc

Highcooley, 2024-06-21:
Same on Linux. Checking the template Folder, e.g. Bambulab folder is completely missing.

rkaneknight, 2024-07-07:
I'm seeing the same issue.  If anyone else sees this when searching for a solution, I found that I can still add an additional printer if I click on the left side printer dropdown and "Select/Remove Printers (system presets)".  At least this dialog works as expected.

duesk, 2024-07-14:
same on Mac

Roofstalker, 2024-07-18:
Same issue on Win 11

HAndreasH, 2024-07-19:
> I'm seeing the same issue. If anyone else sees this when searching for a solution, I found that I can still add an additional printer if I click on the left side printer dropdown and "Select/Remove Printers (system presets)". At least this dialog works as expected.

something seems to be wrong here with database query
i think we have to be patient until someone finds the time to fix the problem
until then we can easily use the work around

shnlmn, 2024-09-02:
Just downloaded OrcaSlicer_Windows_Installer_V2.1.1.exe and am having the exact same issue.

HAndreasH, 2024-09-02:
However, the bug has still not been fixed in the current developer build.
You can of course add printers as described in the previous [comment from rkaneknight](https://github.com/SoftFever/OrcaSlicer/issues/5499#issuecomment-2212608757) already

Stormwing01, 2024-09-28:
Still not fixed

rtm7777, 2024-10-03:
Tried to connect bambu for the first time, same issue

LukasExemplar, 2024-11-03:
Same issue.

octico, 2024-11-24:
Bug still persists in 2.2.0 
[image]

[image]

octico, 2024-11-24:
> I'm seeing the same issue. If anyone else sees this when searching for a solution, I found that I can still add an additional printer if I click on the left side printer dropdown and "Select/Remove Printers (system presets)". At least this dialog works as expected.

Confirmed this workaround is valid.

idoraz1, 2024-11-24:
hey guys! just solved it!. i am using orca slicer 2.1.0
choose "Select/Remove Printers (system presets)" then add the printer you want and then close and open orca. 
now the printer profile should be available.
 
[image]
[image]

MetonIvictus, 2024-12-22:
I am having the same issue, i cannot use orca untill this is fixed. Any ETA on the fix?

idoraz1, 2024-12-22:
> I am having the same issue, i cannot use orca untill this is fixed. Any ETA on the fix?

Look one comment up bro...

MetonIvictus, 2024-12-22:
> > I am having the same issue, i cannot use orca untill this is fixed. Any ETA on the fix?
> 
> Look one comment above bro...

This does not work for me, i have tried this before i posted.

ignis32, 2025-04-13:
version 2.3

Looks like that bug is still there.

Cannot create a printer with non standard nozzle because of that.

HAndreasH, 2025-04-14:
confirm, the bug is still present

[image]

homeland3d, 2025-04-17:
This bug is always present on 2.3 release fresh install --> create new printer --> select bambulab from producer and than come up the pop-up tell the model is not found for this producer please re-select the producer. Sorry my screen is in italian if you need I'll create in english with version 2.3

drakeg77, 2025-05-15:
2.3 release crashes when you click "select/remove printer"
the same issue occurs when trying to add custom printers and bambu

TerseHickory, 2025-12-12:
Still having this issue

PorcelainMouse, 2026-05-17:
Update: I built my own from src. That binary is much more reliable, faster too.  Thank you to the OrcaSlicer team for making good OSS, even if you aren't hosting pre-built RPM pkgs.  You rock, many thanks.