OrcaSlicer/OrcaSlicer #5507: tronxy x5s profile is borken

daef, 2024-05-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Pop!_OS 22.04 LTS

### Printer

Tronxy x5s

### How to reproduce

Try to print *anything* with a temperature other than 205 °C

### Actual results

Prints with 205 °C.

### Expected results

Prints with a temp other than 205 °C

### Project file & Debug log uploads

The whole starting gcode looks really wonky.
It seems way too long and does a lot of stuff no one asked for.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

cochcoder, 2024-05-30:
Do you mean the X5SA 400?

daef, 2024-05-30:
Is there no "default" template, which should suffice for 90% of marlin printers?

I've thrown together this now, which - so far - seems to work:

``
M104 S[nozzle_temperature_initial_layer] ; start heat nozzle
M140 S[bed_temperature_initial_layer]    ; start heat bed
G90    ; abs coords
M83    ; extrude relative
G28    ; home
M190 S[bed_temperature_initial_layer]    ; wait for bed temp
M109 S[nozzle_temperature_initial_layer] ; wait for nozzle temp
G1 X10.1 Y20 Z0.28 F5000.0        ; purge startpos
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ; 1st purge line
G1 X10.4 Y200.0 Z0.28 F5000.0     ; sidestep
G1 X10.4 Y20 Z0.28 F1500.0 E15    ; 2nd purge line
G1 Z2.0 F3000           ; move Z up
```

cochcoder, 2024-05-30:
> Is there no "default" template, which should suffice for 90% of marlin printers?
>
> I've thrown together this now, which - so far - seems to work:
>
> ```
> M104 S[nozzle_temperature_initial_layer] ; start [...]

If you can do some more testing just to make sure that everything works properly, I'll add it to a PR that I currently have.

cochcoder, 2024-06-15:
@daef Do you have any updates on this?

daef, 2024-06-18:
works for me so far - using the profile since

cochcoder, 2024-06-18:
@daef, last question, did you need to change anything else, like relative e distances?

daef, 2024-07-01:
nope

PR #5752 (2024-07-01): Allow variable temperatures with the Tronxy X5SA 400
Fixes #5507