OrcaSlicer/OrcaSlicer #5514: Blank screen on display and preview tab

ALTC-AmLearningToCode, 2024-05-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

windows 11

### Additional system information

Cpu: AMD ryzen 7 5700G with radeon graphics (not using this for graphics) GPU: AMD radeon rx 7600
have 32 gb of ram.

### Printer

Ender 3 v3 KE

### How to reproduce

Open orcaslicer. click "prepare" tab or "preview" tab.

### Actual results

Shows blank, printer settings and everything else is showing, just not the build plate with the model.

### Expected results

Slicer shows build plate and sliced model.

### Project file & Debug log uploads

File name: "crash_Thu_May_30_16_02_06_0"

Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x76c57b58 
Context :0000003C36721A00 
ContextFlag : 0x10005f, EFlags: 0x10246 
Fault address:  0x21BB88B7 0x1:0xA978B7 C:\Program Files\OrcaSlicer\OrcaSlicer.dll

Show CallStack:
7ffc21bb88b7:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc21ca968f:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc21afdbb3:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc21ac452f:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc22c24cfc:hid_write [Unknown file][-1]
7ffc22c2445b:hid_write [Unknown file][-1]
7ffc22c2471b:hid_write [Unknown file][-1]
7ffc22c2a0a7:hid_write [Unknown file][-1]
7ffc22c6a128:hid_write [Unknown file][-1]
7ffc22c6a31a:hid_write [Unknown file][-1]
7ffc22c29a41:hid_write [Unknown file][-1]
7ffc22c2d707:hid_write [Unknown file][-1]
7ffc22c90708:hid_write [Unknown file][-1]
7ffc21aa92ee:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc21223333:Unknown [Unknown file][-1]
7ffc21282484:orcaslicer_main [Unknown file][-1]
7ff7969c5a45:Unknown [Unknown file][-1]
7ff7969c5f06:Unknown [Unknown file][-1]
7ffcc943257d:BaseThreadInitThunk [Unknown file][-1]
7ffccafcaa48:RtlUserThreadStart [Unknown file][-1]

Logical Address:
0x21BB88B7 0x1:0xA978B7 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x21CA968F 0x1:0xB8868F C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x21AFDBB3 0x1:0x9DCBB3 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x21AC452F 0x1:0x9A352F C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x22C24CFC 0x1:0x1B03CFC C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x22C2445B 0x1:0x1B0345B C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x22C2471B 0x1:0x1B0371B C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x22C2A0A7 0x1:0x1B090A7 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x22C6A128 0x1:0x1B49128 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x22C6A31A 0x1:0x1B4931A C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x22C29A41 0x1:0x1B08A41 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x22C2D707 0x1:0x1B0C707 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x22C90708 0x1:0x1B6F708 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x21AA92EE 0x1:0x9882EE C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x21223333 0x1:0x102333 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x21282484 0x1:0x161484 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x969C5A45 0x1:0x4A45 C:\Program Files\OrcaSlicer\orca-slicer.exe
0x969C5F06 0x1:0x4F06 C:\Program Files\OrcaSlicer\orca-slicer.exe
0xC943257D 0x1:0x1157D C:\WINDOWS\System32\KERNEL32.DLL
0xCAFCAA48 0x1:0x59A48 C:\WINDOWS\SYSTEM32\ntdll.dll

File name: "debug_Thu_May_30_21_17_09_6608.log"

[warning]	2024-05-30 21:17:09.609952[Thread 0x0000464c]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_profile_3mf
[error]	2024-05-30 21:17:13.080595[Thread 0x0000464c]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-05-30 21:17:13.084595[Thread 0x0000464c]:Unable to open the fileC:\Users\liamr\OneDrive\Documents\OrcaSlicer\resources/calib/tolerance_test/OrcaToleranceTest.stl
[error]	2024-05-30 21:17:15.230293[Thread 0x0000464c]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

[image]

ALTC-AmLearningToCode, 2024-06-01:
I have reinstalled it multiple times, changed my graphics drivers to older versions and nothing works.

closed 2024-06-01 by ALTC-AmLearningToCode

micartey, 2024-11-25:
I have the same issue @ALTC-AmLearningToCode.
How did you resolve this?

CMD532, 2024-12-26:
Hello, I have the same problem, anyone know how to fix it?

ALTC-AmLearningToCode, 2025-01-05:
My issue seemed to have resolved itself, and I am not sure what caused the original issue. Was MIA due to printer being inoperable but have come back to orcaslicer working properly.