OrcaSlicer/OrcaSlicer #5515: Prime tower doesnt honer size specified

Prutsium, 2024-05-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Win11

### Printer

Vcore4 IDEX

### How to reproduce

See attached .3mf

### Actual results

When specifying a prime tower of 10mm with a purge volume of 3mm3 it creates a rather large prime tower (see image)
[image]

When the 10mm parameter is set to 2 it honors the requested size
[image]

### Expected results

A prime tower of 10x10mm

### Project file & Debug log uploads

[VC4-300-IDEX-ORCA-Cubes.zip](https://github.com/user-attachments/files/15512179/VC4-300-IDEX-ORCA-Cubes.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

hiltpatr, 2024-06-09:
I too have noticed the size of the prime tower being very inconsistent. I try to shrink the width to 10mm on the snapmaker J1 and the prime tower goes thin until I slice the model or try to adjust the width again. Then the tower stretches out across the build plate. I am trying to cut down the prime tower size for Idex printing where you don't need as much purge.

Prutsium, 2024-06-09:
Found the issue and its not the slicers fault. Reduce purge volume for
color changes.

MIIB1, 2024-07-08:
me to same problem

traxman25, 2024-09-04:
The prime tower is using purge volume settings from single nozzle multi color printing. However this setting should be disabled when using a prime tower with multi tool printing as each tool/filament has a setting for it's own prime volume.

MIIB1, 2024-09-04:
Hello,

So for my Snapmaker A350 dual extruder should I disable it correct?

Thank you

closed 2024-12-11 by github-actions[bot]