OrcaSlicer/OrcaSlicer #5517: Prepare window "vibrating" icons

J3LS, 2024-05-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1 beta

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.3 Virginia base: Ubuntu 22.04 jammy

### Additional system information

Laptop System: Dell product: Latitude E7440, Intel Core i7-4600U, Intel Haswell-ULT Integrated Graphics 4Mb 1920x1080 default display - used on this setting, 16Gb ram
Kernel: 5.15.0-107-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Xfce 4.18.1
tk: Gtk 3.24.33 info: xfce4-panel, plank wm: xfwm 4.18.0 vt: 7 dm: LightDM 1.30.0

### Printer

Ender 3 V3 KE

### How to reproduce

When the mouse is moved the toolbar icons above the printer bed preview on the right side of the prepare window vibrate horizontally so fast as to almost be a blur. It appears that the whole icon bar is zooming in and out fast to give this impression. It appears that the top side of the icon bar stays stationary and the sides and bottom of the bar expand/ contract. It's hard to estimate but I would say around 2.5mm at either side and proportionally less at the bottom on my 14in laptop screen. The "vibration" happens only when the mouse is moved and stops immediately the mouse stops.
I tried every display and mouse setting adjustment with no change however **_if the dividing vertical border between the left "Printer" panel and right bed preview panel is moved horizontally the vibration cab be made to stop_** when the mouse moves.
There seems to be "sweet spots" horizontally where the vertical border can be placed where the window behaves normally. On re-opening the app the window behaves as it was when it was last closed.

### Actual results

When the mouse is moved the toolbar icons above the bed on the right side of the prepare window vibrate horizontally so fast as to almost be a blur.

### Expected results

The window should be stable at all times.

### Project file & Debug log uploads

There was no project file loaded. This happened immediately on opening the app after downloading and is the same no matter what operations are in progress

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

I have Orca 2.0. No problems at all so this is specific to 2.1. Both versions are appimages. Both are single files run directly from separate folders with the same parent folder. My system is set for automatic updates.

Last log - quite short:
[error]	2024-05-31 14:25:15.749894[Thread 0x00007f2722bacc80]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2024-05-31 14:25:16.292452[Thread 0x00007f2722bacc80]:02:25:16 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-05-31 14:25:16.297053[Thread 0x00007f2722bacc80]:02:25:16 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-05-31 14:25:16.357521[Thread 0x00007f2722bacc80]:02:25:16 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-05-31 14:25:16.436403[Thread 0x00007f2722bacc80]:02:25:16 PM: Warning: Error running JavaScript: Unsupported result type
[error]	2024-05-31 14:25:18.762715[Thread 0x00007f2722bacc80]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2024-05-31 14:25:18.812757[Thread 0x00007f2722bacc80]:get_version, get_version not supported,return 00.00.00.00!
[error]	2024-05-31 14:25:45.111519[Thread 0x00007f2722bacc80]:calc_exclude_triangles:Unable to create exclude triangles

aledillo, 2024-06-19:
Today I've the same problem for the first time, both with macbook trackpad (mouse off) and with logitech mx3 mouse on. No problem on BambuStudio

rtideas, 2024-07-10:
Same issue for me in 2.1 and beyond. Happens in a lower 1280x720 resolution. If I collapse the side menu, it stops.

Older iMac. Older eyes, hence the lower resolution. 

Seems is shakes when it's in a smaller space. Higher res, is fine.

https://github.com/SoftFever/OrcaSlicer/assets/4334269/6c834cc4-3fa0-4fed-8c6c-6bf47ccfc3f1

shashachu, 2024-07-21:
+1 also seeing this on a Macbook Air. New to Orca so not sure if this is new. I'm on 2.1.1.

SimEyeSee, 2024-08-09:
This has been occurring for me for a few months as well. I have 2.1.1. When the mouse moves around the canvas, it jitters. I can make it stop or start by adjusting the width of the application, or by making the left panel smaller or larger. If the panel is hidden, it also stops.  To make it start again, it seems like the application window has to be a certain width (or range of widths). When the application window is not the problematic width, I can't make the jittering start no matter how wide I make the side panel. But when the application is in the problematic width, I can start the jittering by changing the side panel to a certain width and stop it by making the side panel larger or smaller.

closed 2024-11-15 by github-actions[bot]